riscv-non-isa/riscv-iommu #429: Question about SPA exceeding the maximum address specified by PAS

viktoryou, 2024-09-24:
After the translation of IOMMU, a SPA may exceed maximum address specified by physical address size. From the model, I found a different behavior between one-stage and two-stage translation, see the [code](https://github.com/riscv-non-isa/riscv-iommu/blame/main/iommu_ref_model/libiommu/src/iommu_second_stage_trans.c#L191). The IOMMU model would raise an access fault for the over range of SPA in two-stage translation while leaving the case into PMA/PMP check in one-stage translation. Should the behavior be different?
From my undertstanding, the over range of SPA is regarded as a violation of PMP/PMA check, right?

ved-rivos, 2024-09-26:
Two stage address translation is always active. Either stage may be set to `Bare` to effectively disable the translation and address protection.  However, there is no case where only first stage translation is active. 

The IOMMU is not required to cache all invalid addresses in its address translation caches. A SPA beyond maximum physical address width is not specifically a PMP check. The PMP specified protected memory regions and such protected memory  region checks are done outside the IOMMU. A general PMA check about the attributes of the memory region being compatible with the access type are also done outside the IOMMU. The over range of SPA could be  considered  a special case of a PMA check i.e. these are vacant PMAs. However its a special case because when a range can be addressed by the machine because the address of the range is within the maximum physical address width, some parts of that addressable range may also be vacant and those checks occur outside the IOMMU. The PA size being specified by the IOMMU is to provide the implementation the option not to have to implement caching of invalid SPA in its address translation caches.

viktoryou, 2024-09-27:
> The PA size being specified by the IOMMU is to provide the implementation the option not to have to implement caching of invalid SPA in its address translation caches.

The model would also report an access fault to FQ caused by the invalid SPA. However I could not find a reference in the IOMMU spec. Should we expect such a fault record?

> Two stage address translation is always active. Either stage may be set to Bare to effectively disable the translation and address protection. However, there is no case where only first stage translation is active.

Sorry for my misleading description. Let me rephrase it:
When first stage translation is not-bare and the second stage translation is bare, as well as both stage translation is bare, where should we check whether the SPA is valid or not?

ved-rivos, 2024-09-27:
Actually I was mixing something else in my head when posting in this issue. I see the specific case you are talking about at the link you originally posted. That should be removed. Fixed in #433

yanhe234, 2024-10-31:
Hello sir, there may be a problem with my description. What I want to express is that the high 11 bits of iohgATP and GPA are combined together, but only 4KB was actually used here because the low two bits of iohgATP are 2'b00. Looking forward to your reply

ved-rivos, 2024-10-31:
The bits 43 to 0 of the `iohgatp` provide a 44 bit PPN i.e. a 56-bit page address. The low 2 bits of the PPN are expected to be set to `00 as the root page of G-stage page table must be aligned to 16 KiB. The high 11-bits of the GPA - in case of Sv59x4/Sv48x4/Sv39x4 - are the index into this root table - total of 2048 entries - each entry being 8 bytes - for a total of 16 KiB.

yanhe234, 2024-10-31:
I understand what you said, so my question is why the GPA must be two more bits in the G phase, and then set to 00 in the lower two bits of iohgatp, so I understand that the mapping range has not increased

ved-rivos, 2024-10-31:
I dont understand the question. When G-stage is Sv48x4, the GPA is 50 bit wide. What is held in iohgatp is not a GPA - its a SPA -and the SPA of the G-stage root table is 56-bit wide. The top 11 bits of the 50-bit wide GPA in this example index into the root table.

yanhe234, 2024-10-31:
Thank you very much for your patient answer. I know that I have entangled myself in a meaningless question.