riscv-non-isa/riscv-iommu #430: quetions about iofence.c behavior

leBsky, 2024-09-25:
1. For command iofence.c, if  bit[10] av is 1 and ATS command timeouts before this iofence.c command, then no write to memory should occur. Is it correct?

2. For command iofence.c,  if bit[10] av is 1 and iofence completes without any timeout happens, then the command writes the memory. However, if the write is denied by IOPMP or other error occurs, should the cqh still advance by 1 in this case?

ved-rivos, 2024-09-26:
> For command iofence.c, if bit[10] av is 1 and ATS command timeouts before this iofence.c command, then no write to memory should occur. Is it correct?

When the command times out, it does not complete. The write to memory occurs only if the command completes.

> For command iofence.c, if bit[10] av is 1 and iofence completes without any timeout happens, then the command writes the memory. However, if the write is denied by IOPMP or other error occurs, should the cqh still advance by 1 in this case?

If a memory access made by the command itself leads to a memory fault then the command does not complete and instead the command queue stalls with `cqcsr.cqmf` set to 1. When `cqmf` is 1, the `cqh` references the command in the CQ that caused the error.

leBsky, 2024-09-27:
> When the command times out, it does not complete. The write to memory occurs only if the command completes.
>
>
> If a memory access made by the command itself leads to a memory fault then the command [...]

Clearly. Thanks

closed 2024-09-27 by leBsky

yanhe234, 2024-10-11:
Hello sir, currently I am looking at the smmu code for ARM, and it seems that the command timeout is reported through the PCIe side. smmu itself does not have this timeout report. If there is an error, please correct me. How to implement this according to your understanding by setting the cmdto register under RV? How is timeout determined?

ved-rivos, 2024-10-11:
The timeout value is specified by the PCIe specifications. Could you clarify more what you mean by "reported through PCIe side"

yanhe234, 2024-10-11:
My understanding is that when RC considers the ATC failure of EP to be incomplete, it will report an interrupt, but smmu did not report the interrupt

ved-rivos, 2024-10-11:
Please lookup CERROR_ATC_INV_SYNC and CMD_SYNC.

yanhe234, 2024-10-11:
Hello sir, let me rephrase it as ATS INVAL is the only possibility of timeout. Can we understand it this way？

ved-rivos, 2024-10-11:
Please see section 4.1.2:
> In this version of the specification, only ATS.INVAL command is specified to have a timeout.