riscv-non-isa/riscv-iommu #434: Data corruption in PCIE ATS handling

leBsky, 2024-10-08:
In spec chapter 2.6, the PCIe ATS translation response is generated differently based on the fault. However, conditions like DDT/PDT/ MSI PT data corruption /First Second stage PT data corruption are not considered here. 

Similarly, chapter 2.7, DDT data corruption is not listed.

Please help add those conditions.

ved-rivos, 2024-10-09:
Section 8.4 provides the guidelines on handling RAS errors and specifies a completer abort (CA) as the response.

leBsky, 2024-10-10:
> Section 8.4 provides the guidelines on handling RAS errors and specifies a completer abort (CA) as the response.

For PCIe ATS translation request, a completer abort(CA) response can be generated. However, for PCIe ATS page request, the response can only be invalid request or response failure. So what response should be generated here for PCIe ATS page request?

ved-rivos, 2024-10-10:
For a Page Request, - Invalid Request - is appropriate.

closed 2024-10-11 by leBsky