riscv-non-isa/riscv-iommu #439: tlb_t  why  g_stage no pbmt ?

asdvvcc123, 2024-10-24:
hi,sir: 
why in the tlb_t has vs pbmt ， but no  g_stage pbmt ? 
[image]

if  s1 is bare , s2 is  not bare ，    i think need to use g_stage pbmt;

ved-rivos, 2024-10-26:
This is addressed by the following code. When VS-stage is Bare, the VS-stage PBMT
is by default PMA and hence the G-stage PBMT applies else VS-stage PBMT applies.
This PBMT is what gets cached.
```
   // The page-based memory types (PBMT), if Svpbmt is supported, obtained
    // from the IOMMU address translation page tables. When two-stage address
    // translation is performed the IOMMU provides the page-based memory type
    // as resolved between the G-stage and VS-stage page table
    // The G-stage page tables provide the PBMT as PMA, IO, or NC
    // If VS-stage page tables have PBMT as PMA, then G-stage PBMT is used
    // else VS-stage PBMT overrides.
    // The IO bridge resolves the final memory type
    vs_pte.PBMT = ( vs_pte.PBMT != PMA ) ? vs_pte.PBMT : g_pte.PBMT;

```

asdvvcc123, 2024-10-28:
Thank you very much,  I have another question: why isn't there a walk cache here like in the SMMU to cache the data of every PTW (Page Table Walk)? I want to add a walk cache to the model. so i add the PBMT  in G_stage;  Will there be an addition of a walk cache in the protocol later?

ved-rivos, 2024-10-28:
> hank you very much, I have another question: why isn't there a walk cache here like in the SMMU to cache the data of every PTW (Page Table Walk)? I want to add a walk cache to the model. so i add the PBMT in G_stage; Will there be an addition of a walk cache in the protocol at furture?

Page walk caches and TLBs are not architectural. An implementation is free to implement any form of caching schemes.

> when iommu pbmt = PMA , mean None , So are attributes will be determined by other downstream components , like PMA checker?

PMA means that the attributes specified by the PMA apply. The determination of the attributes is done a downstream component such as a PMA checker.

closed 2024-10-29 by asdvvcc123