riscv-non-isa/riscv-iommu #440: Why there's no need to pay attention to the dc.tc.prpr domain in cases where the page request fails？

summer-LMY, 2024-10-25:
hi.sir:
For the requirements of prpr in section 2.1.3 of the spec, the scenario is as follows: 

When a page request fails to write into the memory's PQ queue and the IOMMU needs to send a page request response, why is it unnecessary to consider the dc.tc.prpr field in the pid domain of the page request response information?

Thanks for your reply in advance.

summer-LMY, 2024-10-25:
Why is it unnecessary to check the dc.tc.prpr field in the page request response information issued by the IOMMU with a failure response?

ved-rivos, 2024-10-26:
if the `prpr` could be determined then it is used. The cases where `prpr` cannot be determined lead to a response failure and in those cases the PRI at the device gets disabled.

summer-LMY, 2024-10-27:
For the two scenarios in spec 3.7 where the IOMMU generates a Response Failure for page request responses and can retrieve the correct dc.tc.prpr, why is it still unnecessary to check prpr?
1. Page-request queue is not enabled (pqcsr.pqen == 0 or pqcsr.pqon == 0)
2. Page-request queue encountered a memory access fault (pqcsr.pqmf == 1)

ved-rivos, 2024-10-28:
The normative reference is:
> When the IOMMU generates the response, the status field of the response depends on the cause of the error. If a fault condition prevents locating a valid device context then the PRPR value assumed is 0.

If the PRPR can be determined then it applies. I see there is some contradicting text on page 41 which needs to be updated to clarify this.

baimengwei, 2024-10-29:
Hi.sir:

if the response code is Response Failure by pq not enable or pq memory fault, the PRPR is can be find in device context. Hope to update it if necessory. Thanks.

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_ats.c#L343-L354

```
    if ( response_code == INVALID_REQUEST || response_code == SUCCESS ) {
        if ( PRPR == 1 ) {
            prgr.PV = pr->PV;
            prgr.PID = pr->PID;
        } else {
            prgr.PV = 0;
            prgr.PID = 0;
        }
    } else {
        prgr.PV = pr->PV;
        prgr.PID = pr->PID;
    }
```
@ved-rivos

ved-rivos, 2024-11-01:
I researched this a bit more. 

The specification states:
> For IOMMU-generated "Page Request Group Response" messages that have status Invalid Request or Success, the PRG-response-PASID-required (PRPR) bit when set to 1 indicates that the IOMMU response message should include a PASID if the associated "Page Request" had a PASID. For IOMMU-generated "Page Request Group Response" with response code set to Response Failure, if the "Page Request" had a PASID then response is generated with a PASID.

This is consistent with the model. 

While the two cases you highlighted, the page queue is not enabled or encountered a memory fault could have been treated differently for the response failure case, the decision was not to special case those two failure conditions. 

Please post if there are further questions.

closed 2024-11-01 by ved-rivos