riscv-non-isa/riscv-iommu #442: when bare no need to check  PAS ?

asdvvcc123, 2024-10-29:
hi  ,sir :

in the tran function ,  when bare ,  jump to step 8  ,but in step 8 ,no code to check PAS;     

[image]
[image]

thanks!

ved-rivos, 2024-10-29:
> in the tran function , when bare , jump to step 8 ,but in step 8 ,no code to check PAS;

The physical address that is the result of an address translation is provided to the IO bridge and further checking of validity of that PA is done by downstream components such as the PMA and PMP checkers.

asdvvcc123, 2024-10-30:
[image]
Hi,Sir: 
 i mean this parameter .    the final PA  no need to  limit by  this  parameter when bare ?   

thanks!

ved-rivos, 2024-10-30:
> i mean this parameter . the final PA no need to limit by this parameter when bare ?

This parameter determines what the IOMMU itself can address. It does not affect the final PA (even when IOMMU is not in Bare mode).

asdvvcc123, 2024-10-31:
ok  ,understand , thank !

closed 2024-10-31 by asdvvcc123