riscv-non-isa/riscv-iommu #443: Does IOMMU need to check the ddtp.iommu_mode of inbound invalid completion?

dongxieDD, 2024-10-30:


dongxieDD, 2024-10-30:
Hello, I have an additional question. When g_reg_file.ddtp.iommu_mode is set to off or bare, how should the IOMMU handle receiving invalid completion messages?

ved-rivos, 2024-10-30:
The iommu should handle invalidation completions normally even when it is in off or bare mode.

ved-rivos, 2024-10-31:
> Thank you for your reply. I have another question: in off mode, should an
> invalidation completion message be reported as a fault (cause=256)?

No. It should not lead to a fault .

dongxieDD, 2024-10-31:
Thank you for your reply.

closed 2024-10-31 by dongxieDD

18772820305, 2024-10-31:
When the iommu_mode switches to off, is it possible for there to be inflight transactions within the IOMMU?

ved-rivos, 2024-10-31:
> When the iommu_mode switches to off, is it possible for there to be inflight transactions within the IOMMU?

When the `iommu_mode` field is changed to `Off`, there may be in-flight transactions in the IOMMU. The IOMMU however guarantees that in-flight transactions, observed at the time of the write to this field, will either be processed using the configurations applicable to the old value of the `iommu_mode` field or be aborted. See section 6.5

yanhe234, 2024-11-05:
What is the process of handling cmd queue when ddtp.mode==bare or off without distinguishing domains? In other words, the processing of commands does not require attention to ddtp.mode? I understand that it is not necessary.
ARM's SMMU distinguishes between secure and non secure domains, while RV currently distinguishes between four domains. I am not sure how the queue processing of cmd needs to focus on ddtp when considering four domains under RV?
I don't know if I described it clearly. Looking forward to your reply

ved-rivos, 2024-11-06:
Could you explain what you mean by four domains?

yanhe234, 2024-11-06:
[image]
This is from the RISC-V Supervisor Domains Access Protection document，Sir

ved-rivos, 2024-11-06:
While that picture depicts 4 domains, the supervisor domain architecture is not limited to 4 domains. The supervisor domains architecture supports instantiation of multiple supervisor domains. One or more of those supervisor domains may be IO capable and may have an IOMMU associated with them. When an IOMMU is associated with a supervisor domain, that supervisor domain has exclusive access to that IOMMU - including its command queues. For further details see:
https://github.com/riscv/riscv-smmtt/blob/main/chapter6.adoc

[image]

yanhe234, 2024-11-07:
> What is the process of handling cmd queue when ddtp.mode==bare or off without distinguishing domains? In other words, the processing of commands does not require attention to ddtp.mode? I understand that it is not necessary. 
So, sir, I need to confirm the first question with you. When ddtp. mode==off or bare, will there be any discarding behavior when handling cmd queues?

ved-rivos, 2024-11-07:
> So, sir, I need to confirm the first question with you. When ddtp. mode==off or bare, will there be any discarding behavior when handling cmd queues?

No. There isnt.

yanhe234, 2024-11-07:
many  thanks

18772820305, 2024-11-14:
Hi, sir, When IOMMU_mode is set to off, should ATS invalidation completion be rejected/aborted/dropped?

ved-rivos, 2024-11-14:
> When IOMMU_mode is set to off, should ATS invalidation completion be rejected/aborted/dropped?

The `iommu_mode` does not have affect how ATS invalidation completions are handled. They are handled normally irrespective of the `iommu_mode`.