riscv-non-isa/riscv-iommu #455: what is difference and relationship between PASID and PSCID?

baimengwei, 2024-12-13:
Prerequisite: The same PCIe device.

Assume the following secenerio:

The PCIe device uses **VA1** to initiate privileged memory access and carries the **PASID1** TLP Prefix.
After being sent to the IOMMU, the IOMMU searches for **DC** according to the BDF, and then finds **PC1** according to PASID1.
**ENS1** is 1 according to **PASID1**. The IOMMU get the **VA1->PA1** relationship corresponding to the **PSCID1** and find Global, which is returned to PCIe device. PCIe considers the VA1 to be Global in all PASIDs.

(1) When the PCIe uses **VA1** to initiate privileged memory access and carries the **PASID2** TLP Prefix. Since the mapping relationship about VA1 has been stored before, the mapping from VA1 to PA1 is directly used.

(2) When the PCIe uses VA1 to initiate privileged memory access and carries the **PASID2** TLP Prefix. And assume that the content stored in PCIe has **expired**. After being sent to IOMMU, IOMMU searches for the **same DC** according to BDF. Then it uses the **PASID2** to search for **PC2** and obtains **PSCID2**, and **ENS2** is 0. The IOMMU find that the privileges is not supported, so a page fault will occur.

The above second point should be reasonable, and the first point is because PCIe believes that the PASID is global and is not sent to IOMMU to search for PC. IOMMU cannot control the fault, resulting in inconsistency.

Is there a problem with the above logic, or why does the inconsistency occur, or does the software reverse-constrain the PSCID to a specific PASID, that is a specific PSCID can only correspond to a specific PASID, making the PC unique ?

baimengwei, 2024-12-16:
Thanks, Just for IP verification of IOMMU.

baimengwei, 2025-02-08:
>It is legal for the device to be provided two PASIDs - such as PASID1 and PASID2 - in this example. The PASID1 in this example is a privileged PASID and PASID2 is a non-privileged PASID. In this case the PASID2 has no privilege to access supervisor memory.

Hi, sir

What I mean is that the **PASID2** is also a privileged PASID,  this **PASID2** might hit PCIe's cache or not. If the PASID2 hit PCIe's cache, the **PC2** in the IOMMU will not be accessed, which might be uncontrollable. So what's the relationship between PASID and PSCID?

Another question,

When PCIe send a request without PASID TLP Prefix, it will report error when the Global bit is set in a response, which hint that the request transaction has no PASID but the Global bit is set at response. The reason why the Global bit is set at response is because of the S-Stage transaction about PSCID. However, PCIe can't identify PSCID, instead, Synopsys PCIe VIP identify PASID is not send, the global bit is not allowed. How the Global bit can be used?

ved-rivos, 2025-02-10:
> What I mean is that the PASID2 is also a privileged PASID, this PASID2 might hit PCIe's cache or not. If the PASID2 hit PCIe's cache, the PC2 in the IOMMU will not be accessed, which might be uncontrollable. 

When a request is received with PASID1, the IOMMU must lookup the PDT using PASID1 to locate its PC. Likewise, when a request is received using PASID2, the IOMMU must lookup the PDT using PASID2 to locate its PC. So the lookup of PDT is solely based on the PASID value in the transaction. 

> So what's the relationship between PASID and PSCID?

One or more PASID may map to the same first-stage page table and in that case may be associated with a common PSCID. Consider a process being allocated work queues in two devices D1 and D2. For D1, the process is allocated PASID P1 and for D2 it is allocated PASID P2. There are two process contexts in this example. The PDT of D1 will provide the PC for P1 and PDT of D2 provides the PC for P2. Both PC however point to the same first-stage page table and hence have identical iosatp and an identical PSCID i.e., the PSCID of the process. 

> When PCIe send a request without PASID TLP Prefix, it will report error when the Global bit is set in a response, which hint that the request transaction has no PASID but the Global bit is set at [...]

For PCIe, the concept of Global, Execute Request, and Privilege Mode Request only applies when the PASID TLP prefix is used. When a ATS translation request is made without a PASID TLP prefix, the completion does not have a PASID TLP prefix. There is no use of Global bit or the Privilege Mode for such devices since they are either not PASID capable or their PASID capability is not enabled.

baimengwei, 2025-02-11:
>When a ATS translation request is made without a PASID TLP prefix, the completion does not have a PASID TLP prefix. There is no use of Global bit or the Privilege Mode for such devices since they are either not PASID capable or their PASID capability is not enabled.

But the c model use vs_pte.G as Global bit to return, that is about PSCID rather than PASID from the aspect of IOMMU . If PASID TLP Prefix is not exist, vs_pte.G is still can be set, which is conflict with PCIe TLP.

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_translate.c#L466

```
        rsp_msg->trsp.Priv   = (req->pid_valid && req->priv_req) ? 1 : 0;
        rsp_msg->trsp.CXL_IO = (req->is_cxl_dev &&
                                ((vs_pte.PBMT != PMA) || (is_msi == 1) || (DC.tc.T2GPA == 1))) ? 1 : 0;
        rsp_msg->trsp.N      = 0;
        rsp_msg->trsp.AMA    = 0;
        rsp_msg->trsp.Global = vs_pte.G;
        rsp_msg->trsp.U      = (is_msi & is_mrif);
        rsp_msg->trsp.R      = (vs_pte.R & g_pte.R);
        rsp_msg->trsp.W      = (vs_pte.W & g_pte.W & vs_pte.D & g_pte.D);
        rsp_msg->trsp.Exe    = (vs_pte.X & g_pte.X & vs_pte.R & g_pte.R) & req->exec_req;
```

baimengwei, 2025-02-11:
>The IOMMU provides the translations to the IO Bridge. It is up to the IO bridge to use the translation information to generate the completion TLP appropriately.

But IOMMU return Global use vs_pte.G to IO Bridge, IO Bridge can't find any information about PASID Global. How should IO Bridge return  the completion TLP appropriately ?

ved-rivos, 2025-02-11:
> But IOMMU return Global use vs_pte.G to IO Bridge, IO Bridge can't find any information about PASID Global. How should IO Bridge return the completion TLP appropriately about Global bit?

We can make that more explicit in the model by masking the Global bit based on whether process_id was valid.

closed 2025-02-12 by baimengwei

dongxieDD, 2025-03-07:
I have a question. The latest C-model has updated the operation of the Global bit regarding "ATS Translation Completion", so that when pid_valid is valid, this Global bit may be 1. However, in some cases, pid_valid is controlled by the IOMMU. For example, when pid_valid = 0, DC.tc.DPE = 1 and vs_pte.G = 1, the IOMMU will set pid_valid to 1. At this time, there will still be a TLP conflict in PCIE.
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_translate.c#L212C1-L218C6
    if ( DC.tc.DPE == 1 && req->pid_valid == 0 ) {
        req->pid_valid = 1;
        req->process_id = 0;
        req->priv_req = 0;
    }
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_translate.c#L473
        rsp_msg->trsp.Global = req->pid_valid && (is_msi == 0) ? vs_pte.G : 0;

ved-rivos, 2025-03-07:
@dongxieDD Thanks for reporting that. This is fixed by #521

HDntown, 2025-03-11:
Here I have a question: If our design or system requires sending pid_valid back to the PCIe side, is this operation reasonable, since pid_valid originally comes from PCIe? If it is reasonable, should it return 1 or 0 in the following scenarios?

`if ( DC.tc.DPE == 1 && req->pid_valid == 0 ) {
req->pid_valid = 1;
req->process_id = 0;
req->priv_req = 0;
}‘

ved-rivos, 2025-03-11:
You would want to use the pasid_valid and not pid_valid

HDntown, 2025-03-12:
How to understand it. Does it mean to return pasid_valid instead of pid_valid?

ved-rivos, 2025-03-12:
If the request had a pasid then its value is held in pasid_valid. You can use that to determine if the original request had a pasid