riscv-non-isa/riscv-iommu #456: Is iocountovf set somewhere?

baimengwei, 2024-12-16:
iocountovf[1:31] is a shadow of iohpmevt[1:31].of, but in c model, iocountovf is not be set in anywhere.

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_hpm.c#L101

ved-rivos, 2024-12-16:
See https://github.com/riscv-non-isa/riscv-iommu/blob/666192e15d7e6fb5c591a7935690fde274570d5a/iommu_ref_model/libiommu/src/iommu_reg.c#L42

ved-rivos, 2024-12-16:
> but in c model, iocountovf is not be set in anywhere.

The bits in this register are read-only shadows of the `OF` bits in `iohpmcycles/iohpmevt*`. Bit 0 of `iocountovf` is a read-only shadow of the OF bit in `iohpmcycles` and bits 31:1 are read-only shadows of the `OF` bit in the corresponding `iohpmevt` registers. There is no explicit "set" of bits in `iocountovf` as this register does not have any storage of its own.

closed 2024-12-17 by baimengwei