riscv-non-isa/riscv-iommu #460: Scenarios for triggering a fault report message with cause=264

summer-LMY, 2024-12-31:
Under what circumstances will IOMMU report a fault report message with cause=264?

ved-rivos, 2024-12-31:
Cause code 264 is reported when the IOMMU encounters an access fault when attempting to access the memory resident interrupt file (MRIF).

summer-LMY, 2024-12-31:
Thanks for the answer, I have one more question，For MSI in MSI_FLAT mode with access fault, why IOMMU has no corresponding fault report meesage?

summer-LMY, 2024-12-31:
Thanks and best regards.

jjmuns, 2025-01-09:
I have another question. If notice msi has an Access fault when trying to access memory, what will IOMMU do? Does IOMMU need to report the fault?

ved-rivos, 2025-01-09:
> If notice msi has an Access fault when trying to access memory, what will IOMMU do?

The IOMMU will report a fault with cause code = 273 ("IOMMU MSI write access fault").

jjmuns, 2025-01-10:
thanks，get it

summer-LMY, 2025-01-10:
Another question, when an access fault occurs when IOMMU writes an msi_flat message to IMSIC，does IOMMU need to report a fault report message and what is the cause of the fault report message?

ved-rivos, 2025-01-10:
> Another question, when an access fault occurs when IOMMU writes an msi_flat message to IMSIC

A MSI is a memory write transaction. The IOMMU provides address translations to the MSI write - in flat mode, or MRIF mode, or when MSI page tables are not active. Completion of the memory write transaction using the translated address provided by the IOMMU is done by the IO bridge.

jjmuns, 2025-01-14:
When IOMMU sent a notice msi, an access fault appeared. At this time, if IOMMU wants to report the fault report with cause=264, how to configure the contents of its fault report, such as ttyp, iotval ，did ，pid， pid valid, etc?

ved-rivos, 2025-01-17:
> At this time, if IOMMU wants to report the fault report with cause=264, how to configure the contents of its fault report, such as ttyp, iotval ，did ，pid， pid valid, etc?

These should be determined by the memory write request that was determined to a MSI write.

closed 2025-01-25 by ved-rivos

viktoryou, 2025-11-28:
Thanks for the explanation. But why the cause is set to 273 instead of 264?

viktoryou, 2025-12-01:
When access fault happened in either of the process below and would be reported by an implementation, from my understanding:
(1) recording incoming MSI in an MRIF 
     -> cause set to 264,  ttyp set to 3, iotval set to the IOVA, iotval2 set to 0,  DID, PV, PID, and PRIV fields set according to the inbound transaction
(2) generating associated notice MSI 
     ->   cause set to 273, ttyp set to 3, iotval set to the IOVA, iotval2 set to 0, DID, PV, PID, and PRIV fields set according to the inbound transaction
Since _A translation request  from  a  device  to  a  GPA  that  is  mapped  through  a  MRIF  mode  MSI  PTE  is  not eligible  to  receive  a  translated  address._ The inbound transaction would always be untranslated, then ttyp would never be set to 7. 
Is that right?

viktoryou, 2025-12-02:
Thanks for the help.