riscv-non-isa/riscv-iommu #463: Question about software guidelines for invalidations

zhuotianshu, 2025-01-11:
Hi!

In the spec:

> If software changes a leaf-level DDT entry (i.e, a device context (DC), of device with device_id = D)
> then the following invalidations must be performed:
> ⚫ IODIR.INVAL_DDT with DV=1 and DID=D
> ⚫ If DC.iohgatp.MODE != Bare
> ⚫ IOTINVAL.VMA with GV=1, AV=PSCV=0, and GSCID=DC.iohgatp.GSCID
> ⚫ IOTINVAL.GVMA with GV=1, AV=0, and GSCID=DC.iohgatp.GSCID
> ⚫ else
> ⚫ If DC.tc.PDTV==1
> ■ IOTINVAL.VMA with GV=AV=PSCV=0
> ⚫ else if DC.fsc.MODE != Bare
> ■ IOTINVAL.VMA with GV=AV=0 and PSCV=1, and PSCID=DC.ta.PSCID
> If software changes a non-leaf-level DDT entry the following invalidations must be performed:
> ⚫ IODIR.INVAL_DDT with DV=0

What is the reason for clearing the PDT and page table cache when modifying a leaf DDT entry, and why is it unnecessary to do so when modifying a non-leaf DDT entry? Thanks.

ved-rivos, 2025-01-11:
Some contents of the device context are used as tags when first-stage page table entries are cached. Likewise the contents of the device context are used to tag entries cached from the second-stage page tables. Some implementations may also employ an address translation cache that maps GVA to SPA and not maintain separate caches for first and second stage page table entries. The PDT entries hold a GPA and when the PDT entries are cached, some implementations may cache in the PDT caches the SPA obtained using the second stage page tables pointed to by iohgatp in the device context. Due to these reasons the PDT and page table caches should be invalidated when a leaf DDT entry is modified. On the other hand the non-leaf entries are only used to locate a device context and thus if a non-leaf entry is modified the DDT caches need to be invalidated. Note that if a non-leaf entry is modified such that the leaf entry contents before and after the modification are not identical then this is also considered a modification of the leaf DDT entry.

zhuotianshu, 2025-01-12:
Thank you for replying!

> Some contents of the device context are used as tags when first-stage page table entries are cached. Likewise the contents of the device context are used to tag entries cached from the second-stage page tables.

Yes, the contents of GSCID and PSCID, along with other information such as the Sv-mode(Sv39, Sv 48, etc.), are stored as TLB tags. Conceptually, before a TLB lookup, the DDT cache is first checked to retrieve this information, which is then used to compare with the TLB tags.

However, since these pieces of information are stored as tags (instead of data), it is not possible to mistakenly hit these entries when modifying the DC, followed by invalidating the DC cache. In this regard, they seem similar to the VMID and ASID in HARTs. The purpose of VMID/ASID is that when modifying satp/hgatp, the TLB does not need to be invalidated. Similarly, when modifying the DC (which is conceptually similar to modifying satp/hgatp), I would expect that invalidating the TLB would also be unnecessary.

However, unlike in HARTs, there may not be “context switching” in the IOMMU. Modifying GSCID and PSCID indicates that the previous task is fully complete, and thus the corresponding TLB entries can be cleared since they won’t be used again. Is this an implicit condition for modifying the DC?

ved-rivos, 2025-01-12:
The DC also holds the page table pointers - hgatp and iosatp/pdtp - and contents of these page tables are cached as data in the TLB. Changes to these page table pointers also necessitate invalidation of the cached contents. Unlike a CPU MMU, the DC is not context switched as all contexts are active at all times. Unlike the CPU where a memory access only gets generated by the current context executing on that hart, the devices may generate a memory access at any time. Hence its not generally possible to context switch a DC. While some changes to a DC may not require a invalidation, the guidelines are not trying to do it surgically as modifications of the DC are expected to be infrequent and the guideline is for the common use cases.

zhuotianshu, 2025-01-13:
Got it, now I understand the purpose of the invalidation sequences. Thank you!

closed 2025-01-13 by zhuotianshu

viktoryou, 2025-09-26:
I am confused about the above explanation and I quoted here：
> Note that if a non-leaf entry is modified such that the leaf entry contents before and after the modification are not identical then this is also considered a modification of the leaf DDT entry.
Changing a non-leaf process or device directory entry both may likely alter the content of DC/PC. From my understanding, TLB that references the previous context maybe outdated. Would it be better if they should be invalidated even they may not be used (by requiring additional iotinval.VMA for example)?

ved-rivos, 2025-09-26:
If the change to a non-leaf entry alters the contents of the DC or PC then the DC or PC needs to be invalidated for the change to visible to the IOMMU. Likewise if that affects the PTEs then an IOTINVAL.VMA and/or IOTINVAL.GVMA is additionally needed to make that change visible to the IOMMU. The guidelines for changing PTE entries cover that aspect and should be considered when deciding the number and types of invalidations required.

viktoryou, 2026-02-09:
I would like to confirm one more related thing here. If software changes a non-leaf-level DDT entry, then _IODIR.INVAL_DDT_  with  _DV=0_ invalidations must be performed. Would this command act like an invalidate-all operation, since all translation cache would be influenced due to the content change of the DC? Or the guidelines for changing PTE entries/PC cover that aspect and software should issue these commands one-by-one?
Thanks in advance.