riscv-non-isa/riscv-iommu #466: RERI error code mappings

zetalog, 2025-01-14:
Should IOMMU specification define RERI error code mappings?

Should cfg cache, ptw cache, tlb each be assigned a different error code than what are defined for CPU internal memories, or should they be mapped to the existing RERI error codes?

ved-rivos, 2025-01-17:
The RERI intent was to define error code encoding for some common types of error codes that are generically applicable to various processing elements such as the CPU MMU, IOMMU, etc. The error record identifies it as being an IOMMU vs. a CPU MMU. So would be appropriate to use the standard Error code encoding for IOMMU RAS error records where applicable - such as those for TLB/page walk caches, internal data path errors, corrupted data access such as attempted consumption of poisoned data by IOMMU on access to its in-memory data structures.

zetalog, 2025-01-21:
For page table walk cache and TLBs, using error code 9 (TLB/Page-walk cache data (e.g., ECC error on TLB data) error) and 10 (TLB/Page-walk cache address/control state (e.g., ECC error on TLB tag) error) seem to be proper.

However, which error code should be applied for DC/PC cache? Do you prefer 17 (Internal datapath, memory, or execution units error (e.g, ALU datapath parity).

ved-rivos, 2025-01-21:
> However, which error code should be applied for DC/PC cache? Do you prefer 17 (Internal datapath, memory, or execution units error (e.g, ALU datapath parity).

One option is to use code 9 and 10 also for DC and PC. Second option is to use code 17. A third option is to use a vendor specific code. In all these cases, the info_i register can be used to provide further information about the structure that encountered the error.

zetalog, 2025-01-22:
Thanks for sharing your opinion, we'll choose one of the options.

closed 2025-01-22 by zetalog