riscv-non-isa/riscv-iommu #467: The condition that writing to ddtp can't be synchronized with writing operation

zhengxuan-luan, 2025-01-14:
Hi

In Section 5.5 the following statement is made:

> A write to ddtp may require the IOMMU to perform many operations that may not occur synchronously to the write. When a write is observed by the ddtp, the busy bit is set to 1. When the busy bit is 1, behavior of additional writes to the ddtp is UNSPECIFIED. Some implementations may ignore the second write and others may perform the actions determined by the second write. Software must verify that the busy bit is 0 before writing to the ddtp.

If IOMMU immediately modifies ddtp register upon observing a write operation and discards any inbound transactions that have not yet begun processing, even while there are still outstanding transactions(e.g.  this transaction is looking up PDT), what consequences might this cause? During the period when this write operation is deferred, what should IOMMU do to avoid this problem?

Thanks!

zhengxuan-luan, 2025-01-17:
> When an IOMMU is turned off, the specification requires that transactions that have been observed by the IOMMU, at the time of write to that field, will either be processed by the old value of the iommu_mode or will be aborted. So its allowed for the IOMMU to either complete the transaction or abort the transaction.

If one IOMMU implementation aborts these transactions when it observes writing to ddtp, its ddtp.busy can be hardwired to 0; otherwise this bit should be set 1 until these transactions have been processed by the old value. Am I correct in understanding it this way?  

Thanks!

ved-rivos, 2025-01-17:
An IOMMU may hold the ddtp busy for implementation defined reasons. For instance, an IOMMU may hold ddtp busy because it has a state machine that needs to execute to perform the necessary actions and such a state machine may need say 50 cycles to execute in which case the busy stays high for 50 cycles. However, the reason an IOMMU may make ddtp.busy is orthogonal to the choice of whether it aborts the transactions or processes them with the old value. It is legal for the IOMMU implementation to not complete all actions it needs to perform synchronous with the write and thereby hold the ddtp busy but also to start aborting transactions as soon as the ddtp write is observed.

ved-rivos, 2025-01-25:
Hope that addressed the question.

closed 2025-01-25 by ved-rivos