riscv-non-isa/riscv-iommu #469: field "WSI" of IOMMU Command-queue Fence commands -> how configure this field to  generate line signal interruption ？

devotee01, 2025-02-12:
In order to generate an interrupt through the WSI field in the command-queue fence, do I need to configure the enable bit in the register cqcsr in addition to setting this WSI to 1?

ved-rivos, 2025-02-12:
An IOMMU fetches and executes commands from the command queue only when the command queue is enabled .

devotee01, 2025-02-12:
> An IOMMU fetches and executes commands from the command queue only when the command queue is enabled .

Hi, iommu fetches and executes commands from the command queue ，when  field "WSI" of fence command is set to 1, then will occuer the wire signaled interrupt ?

ved-rivos, 2025-02-12:
Yes

devotee01, 2025-02-12:
> Yes

In addition to the above description, when an interrupt occurs through the WSI field in fence command, do you need to configure the interrupt enable bit in the cqcsr register ?

devotee01, 2025-02-12:
> Which interrupt enable are you referring to

[image]
wheather need to enable  this field ?

ved-rivos, 2025-02-12:
That bit is the command queue enable bit.

ved-rivos, 2025-02-12:
The cie bit enables interrupts from the command queue. Please also read description of fence_w_ip bit

ved-rivos, 2025-02-13:
Hope that addressed the question. Please ask if there are further questions.

closed 2025-02-13 by ved-rivos