riscv-non-isa/riscv-iommu #470: change msi page table

devotee01, 2025-02-12:
Are there only two scenarios for changing msi page table described in Section 6.3.3 of the manual(GV=AV=1 or GV=1, AV=0; )?Do you need to consider phase 1 and phase 2 types for validating the iotinval.GVMA command to invalid msi page table entry? 
2. When verifying the iotinval.GVMA command, the msi page table needs to be modified first. Is PPN and A[11:0] in the msi page table modified? Because the actual physical address of the conversion is: PPN<<12 | A[11:0]

ved-rivos, 2025-02-12:
I am sorry, I am not able to get the two questions. The section 7.3.3 specifies provides the guideline that if a single entry in the MSI table needs to be invalidated then use the IOTINVAL.GVMA with the address and GSCID for that enrty. And if all entries need to be invalidated then use the form with the address excluded.

ved-rivos, 2025-02-13:
Hope that answered the question. Feel free to ask if there are further questions.

closed 2025-02-13 by ved-rivos