riscv-non-isa/riscv-iommu #471: How are the domains PW and PR in Fence commands validated?

devotee01, 2025-02-12:
If you want to validate domains PW and PR in Fence commands, how do you create incentives and how do you validate this functionality?

devotee01, 2025-02-12:
[image]
spec description follow: 
      The PR bit, when set to 1, can be used to request that the IOMMU ensure that all previous read
requests from devices that have already been processed by the IOMMU be committed to a global
ordering point such that they can be observed by all RISC-V harts and IOMMUs in the system.
     how to understand the "be committed to a global ordering point"? and all RISC-V harts and IOMMUs in the system how to observed global ordering point ?

ved-rivos, 2025-02-12:
A global ordering point relates to a conceptual point in a system that ensures that an operation reaching that ensures that the effects of such operations will be observed by any subsequent operation

devotee01, 2025-02-13:
> A global ordering point relates to a conceptual point in a system that ensures that an operation reaching that ensures that the effects of such operations will be observed by any subsequent operation
 Hi,how to ensures that the effects of such operations will be observed by any subsequent operation? What is the mechanism of being observed ?

ved-rivos, 2025-02-13:
> how to ensures that the effects of such operations will be observed by any subsequent operation?

That is the property of ensuring that all previous reads and writes have made it to the global ordering point.

> What is the mechanism of being observed ?

If PR is set to 1, then all previous reads requests from the device. This ensures that after IOFENCE completes, all previous reads have been observed by all harts and IOMMUs in the system. If PW is set to 1, then all previous write requests from the device. This ensures that after the IOFENCE completes, all previous writes have been observed by all harts and IOMMUs in the system.

devotee01, 2025-02-13:
> That is the property of ensuring that all previous reads and writes have made it to the global ordering point.
>
>
> If PR is set to 1, then all previous reads requests from the device. This ensures [...]
 If PR is set to 1, and  after IOFENCE completes， the hart how to observe all previous reads operation? 

 is there mechanism like a queue that stores all previous reads operation ? 

 or the hart use  the device_id of each device  to observe the previous read request operation ？

devotee01, 2025-02-13:
spec description about PR and PW ， about global ordering point， in what ordering  for read request? 

this ordering is different depend  on access for memmory  or access for io memory ？ thanks!

ved-rivos, 2025-02-13:
> is there mechanism like a queue that stores all previous reads operation ? or the hart use the device_id of each device to observe the previous read request operation ？

These detail vary by  implementation. 

> spec description about PR and PW ， about global ordering point， in what ordering for read request?

If PR is 1, then all read requests prior to IOFENCE have made it to the global ordering point when IOFENCE completes.

> this ordering is different depend on access for memmory or access for io memory ？ thanks!

The ordering is with respect to the IOFENCE. The completion guarantees are provided for accesses to main-memory. For access to IO memory - such as peer-to-peer accesses among devices, the ordering depends on the implementation and the IO protocol.

closed 2025-02-13 by ved-rivos