riscv-non-isa/riscv-iommu #473: addr[63:2] of fence command  how to configure  register seteipnum_le and seteipnum_be  address?

devotee01, 2025-02-13:
[image]
 such as spec description：  Software may configure the ADDR[63:2] command operand to specify the address of
the seteipnum_le/seteipnum_be register ，
 so ADDR[63:2]  How much should be configured? 
 iommu how know this configure is address of seteipnum_le/seteipnum_be register ? 

follow is advance interrupt spec description about seteipnum_le/seteipnum_be:
      
[image]
so， if ADDR[63:2] configure value is 0x000 expressed as address of seteipnum_le register ? 
        if ADDR[63:2] configure value is 0x004 expressed as address of seteipnum_be register ?

ved-rivos, 2025-02-13:
0 ans 4 are offsets of those registers . The address of those registers should be configured .

devotee01, 2025-02-13:
> 0 ans 4 are offsets of those registers . The address of those registers should be configured .

As shown in the picture above, 95:64 value is configured to register seteipnum_le， offset is 0x000？
125:96 value is configured to register seteipnum_le， offset is 0x000？

ved-rivos, 2025-02-13:
If for example your IMSIC interrupt file occupies the address range 0xC0000000 to 0xC0000FFF.  Then its seteipnum_le register is at offset 0 i.e. its address is 0xC00000000 and the seteipnum_be is at offset 4 i.e. its address is 0xC0000004. You may now program ADDR[63:2] with one of these two addresses depending on the endianness.

closed 2025-02-13 by ved-rivos

devotee01, 2025-02-13:
> If for example your IMSIC interrupt file occupies the address range 0xC0000000 to 0xC0000FFF. Then its seteipnum_le register is at offset 0 i.e. its address is 0xC00000000 and the seteipnum_be is at [...]

ok ,so iofence.c completion，there is an external interrupt notification on IOFENCE.C completion?

ved-rivos, 2025-02-13:
If the iofence operands are setup to generate one then yes