riscv-non-isa/riscv-iommu #475: whether use IOTINVAL.VMA/GVMA command to invalid first-stage/second-stage PTE when use iodir.inval_ddt command with dv(dv=0) to invalid ddt,pdt ?

devotee01, 2025-02-13:
[image]
when use  iodir.inval_ddt command to invalidate the ddt cached entry, wheather need to use  IOTINVAL.VMA/GVMA command to invalid first-stage/second-stage PTE during the address translation ? thanks!

ved-rivos, 2025-02-13:
I am not sure I fully understand the question. The commands shown in the guidelines are to invalidate all cached translations from the address translation caches i.e. AV=0.

devotee01, 2025-02-13:
> section shows the types of commands that need to be issued and the conditions that determine the type of commands to be issued

This section shows the types of commands More is IOTINVAL command, topic is “7.3.1. Changing device directory table entry”

devotee01, 2025-02-13:
> Please explain what the question is

hi, when Changing device directory table entry，In addition to use  iodir.inval_ddt command to invalidate the ddt cached entry . whether need to judgment the fsc.mode and iohgatp.mode, Depending on the mode, whether  need to use IOTINVAL.VMA/GVMA command to invalid first-stage/second-stage PTE ?   

this section topic description is about Changing device directory table entry, but use the command include IODIR.INVAL_DDT and  IOTINVAL.VMA/GVMA. so With the above description  confusion about Whether to use iodir.inval_ddt command and IOTINVAL.VMA/GVMA command simultaneously for  Changing device directory table entry in address translation？

ved-rivos, 2025-02-13:
The section states that when changing DDT entry , in addition to invalidating a cached device context the address translation caches may also need to be issued because those caches may also cache some contents from a device context. 

So answer to your first question is Yes

devotee01, 2025-02-13:
> The section states that when changing DDT entry , in addition to invalidating a cached device context the address translation caches may also need to be issued because those caches may also cache some contents from a device context.
> 
> So answer to your first question is Yes

thanks, are you riscv-iommu expert, I look forward to communicating with you if I have any questions later.

closed 2025-02-13 by ved-rivos

devotee01, 2025-02-13:
> thanks, are you riscv-iommu expert, I look forward to communicating with you if I have any questions later.

[image]
for Changing device directory table entry, It is not clear scene about Changing device directory table entry whether  is overall？

devotee01, 2025-02-14:
> These guidelines are intended to be robust. Changing a device context is not a frequent operation and does not need to be hyper optimized . That’s why the guideline does not try to provide guidance in a field by field manner. However these are guidelines and not mandates. If you want to not follow this guideline or make finer grained decisions for your implementation then that may be okay. If you have questions about specific fields and whether you can avoid a iotinval command then please ask about that and we may be able to help with that

[image]
for  above description,
         If DC.iohgatp.MODE != Bare, then second stage address translation is not bare , but does not state first-stage mode type, why use IOTINVAL.VMA to invalidate first-stage PTE .