riscv-non-isa/riscv-iommu #478: how to make  msi mrif fault cause 264？

devotee01, 2025-02-14:
[image]
 description of risc-v iommu about msi mrif fault cause 264 is less,  how to make  msi mrif fault cause 264？ or what is this type fault ?

ved-rivos, 2025-02-14:
The cause 264 occurs when the IOMMU attempt to access an MRIF to set an interrupt pending bit leads to an access violation. Such access violations may occur if the MRIF memory was located in memory that is not accessible to the IOMMU various reasons such as that memory being non-existent or being protected through a memory protection mechanism.

closed 2025-02-15 by ved-rivos

devotee01, 2025-02-18:
> The cause 264 occurs when the IOMMU attempt to access an MRIF to set an interrupt pending bit leads to an access violation. Such access violations may occur if the MRIF memory was located in memory [...]

so, how to set MRIF memory location, is it through MRIF Address[55:9] to set MRIF memory location? as follows: 

[image]

ved-rivos, 2025-02-18:
Yes