riscv-non-isa/riscv-iommu #482: understanding about the function of "invalidation Request"，how to understand this sentence： “Invalidation Request” message is used to clear a specific subset of the address range from the address translation cache in a device function

devotee01, 2025-02-17:
[image]

First, let me explain my understanding of this sentence.  “Invalidation Request” message can clear the translated address in address translation cache of device,  so if pcie device seed the address(have be clear by "invalidation request" handle) translate request， the iommu need to translate this address？ if  pcie device seed the address(address in address translation cache of device and can‘t be clear by "invalidation request" ) , then iommu can't need to translate this address ?

ved-rivos, 2025-02-17:
A PCIe device may obtain translations using a ATS Translation Requests and cache them in the address translation caches (ATC) in the device. When software changes the page tables, it issues an Invalidation Request message with a range of addresses that correspond to the modified entries in the page tables. The purpose of this Invalidation Request is to clear the cached information from those page table entries in the devices address translation cache such that the device will issue a new Translation Request for those addresses. The new Translation Requests will then observe the latest state of those page table entries. The purpose of the Invalidation Request is to synchronize the device ATC with the latest copy of page tables in host memory.

devotee01, 2025-02-17:
> A PCIe device may obtain translations using a ATS Translation Requests and cache them in the address translation caches (ATC) in the device. When software changes the page tables, it issues an [...]

so , if use Invalidation Request（ if Invalidation Request message with a range of addresses is ‘h1000~2000 ）to pcie device, then   those  range of addresses correspond to the modified entries in the page tables will be invalid？

closed 2025-02-17 by ved-rivos