riscv-non-isa/riscv-iommu #483: access fault  could lead these interrupt：cqmf,fqmf,pqmf ，then how to make access fault ? what type is access fault ?

devotee01, 2025-02-17:
description about access fault can lead cqmf,fqmf,pqmf  in spec as follows .  then how to make access fault ? 

[image]

[image]

[image]

closed 2025-02-17 by ved-rivos

devotee01, 2025-02-20:
> The IOMMU may encounter an access fault on access to memory.

> The IOMMU may encounter an access fault on access to memory. A specific case of such an access fault occurs when the IOMMU attempts to access the fault-queue, command-queue, or page-request-queue. When an access fault occurs on access to these queues, the operations of the queues are impacted and the `fqmf`, `cqmf`, and `pqmf` bits are set respectively to indicate that these queues are not operational anymore due to a memory fault being encountered. An interrupt is generated to notify software about this condition. The behavior of the queues when these memory fault conditions occur are as noted in the text you highlighted.

how to make cqmf fault ？

ved-rivos, 2025-02-20:
To fetch commands from the command queue, the iommu needs to access the memory holding the commands . An attempt to access that memory may lead to a fault . Any memory that the iommu needs to access as part of command execution itself may lead to a fault . If faults are encountered on attempts to fetch or execute commands from the command queue then a command queue memory fault (cqmf) occurs