riscv-non-isa/riscv-iommu #490: how to understand cause=1  situation ?

devotee01, 2025-02-19:
[image]
1. why the transaction is an Untranslated or Translated read-for-execute then stop and report "Instruction access fault" ? 
2. how to make this  situation ?

devotee01, 2025-02-20:
> Generating a Translated or Untranslated read-for-execute type transaction with an address that translates to a MSI PTE will lead to this fault.

how to generate the  Translated or Untranslated read-for-execute type transaction ?  Translated  means the address has been translated? Untranslated read-for-execute type will be control what component?

ved-rivos, 2025-02-20:
I am not sure what kind of test environment you are working with to help with suggestions on generating stimulus .

ved-rivos, 2025-02-20:
Specifically look for how to generate a Mrd TLP and how to control the address type AT field or the  TLP and how to control the PASID TLP prefix or OHC - if in flit mode - to set the execute requested bit in the prefix