riscv-non-isa/riscv-iommu #491: HW A/D update when implicit g-stage page table

1252005989, 2025-02-19:
Hi,
in ref model,  If IOMMU HW A/D bit update are enabled the implicit accesses are treated as writes. However, when passing parameters, the value passed for is_write is 0. so D bit is never updated . I want to know how to handle D bit update for implicit g-stage page table.

[image]

[image]

1252005989, 2025-02-19:
I think this expression may change to :
 if ( ( (is_write == 1)|| (is_implicit == 1) )  && (amo_pte.W == 1) ) amo_pte.D = 1;

ved-rivos, 2025-02-19:
Thanks for reporting this. Fixed by #492

1252005989, 2025-02-20:
Thanks. And there is another question. for example:

1252005989, 2025-02-20:
Thanks. And there is another question. for example:
Original  inbound transaction type is write, 
 leaf vs-pte A bit=1, D bit = 0,
implicit g-pte A bit=1, D bit = 0, W=1
DC.SADE=1, DC.GADE=0

I think g-pte D bit needs to be updated but there is no permission (GADE=0), so guest-page-fault will happen.
but in ref model ,g-pte D bit is not updated and not report fault, so may  this expression (GADE == 0)  need to removed?
[image]

ved-rivos, 2025-02-20:
You are right. Updated by https://github.com/riscv-non-isa/riscv-iommu/pull/497

1252005989, 2025-02-21:
Thanks.  Finally I want to confirm that since implicit accesses are treated as writes,  in some cases unexpected implicit guest-page-fault will happen, and this scenario conforms to the design concepts of the reference model or the IOMMU specification/privileged architecture specification.
an example is :
leaf vs-pte A bit=1, D bit = 1,
implicit g-pte A bit=1, D bit = 0, W=1
DC.SADE=1, DC.GADE=0
in this example an implicit guest-page-fault happens(because it wants to update g-pte D bit) but maybe it does not need to happen

ved-rivos, 2025-02-21:
Yes that is as expected

1252005989, 2025-02-21:
Thank you for the explanation.