riscv-non-isa/riscv-iommu #493: for fence command test, whether can once write to command queue more command include Page-Table cache invalidation,  directory cache invalidation,andATS commands?

devotee01, 2025-02-20:
[image]
as spec description, wheather exsit this situation is descripted as topic ？

ved-rivos, 2025-02-20:
If your question is whether you can queue more commands into the queue when a fence command is in progress then answer is yes. You can keep queuing more commands - the iommu will get to them after the iofence command completes

devotee01, 2025-02-21:
> If your question is whether you can queue more commands into the queue when a fence command is in progress then answer is yes. You can keep queuing more commands - the iommu will get to them after the iofence command complete
whether can queues more command(eg:queues command to queue in order: iotinval.vma, iotinval.gvma,inval.ddt,inval.pdt,fence.c,inval.ddt,inval.pdt) to queue?

ved-rivos, 2025-02-21:
Yes

closed 2025-02-21 by ved-rivos