riscv-non-isa/riscv-iommu #498: Question about pte.NAPOT and page size

HDntown, 2025-02-21:
When a PTE has N=1, the PTE represents a translation that is part of a range of contiguous virtual-to-physical translations with the same values for PTE bits 5–0. Such ranges must be of a naturally aligned power-of-2 (NAPOT) granularity larger than the base page size. So we think page size would be change  in this scenario. However, in iommu_two_stage_trans.c and iommu_second_stage_trans.c, we can see if set pte->N 1,  we found that why does the page size not change when pte.N equals 1?

`    if ( pte->N )
        pte->PPN = (pte->PPN & ~0xF) | ((iova / PAGESIZE) & 0xF);

    // The G bit designates a global mapping. Global mappings are those that exist
    // in all address spaces.  For non-leaf PTEs, the global setting implies that
    // all mappings in the subsequent levels of the page table are global.
    pte->G = NL_G | pte->G;

    // The translated physical address is given as follows:
    // pa.pgoff = va.pgoff.
    // If i > 0, then this is a superpage translation and pa.ppn[i − 1 : 0] = va.vpn[i − 1 : 0].
    // pa.ppn[LEVELS − 1 : i] = pte.ppn[LEVELS − 1 : i]
    *pa = ((pte->PPN * PAGESIZE) & ~(*page_sz - 1)) | (iova & (*page_sz - 1));
    return 0;`

ved-rivos, 2025-02-21:
1. Svinval allows but does not require an implementation to cache 64K page size translations . When an implementation does not support 64K page sizes natively it can interpret each napot pte as providing translation for a 4K page determine by the page number encoded by the Napot bits
2. If an implementation supports interpreting Napot pte as 64K pages then it may provide such 64K sized translations in translation responses

HDntown, 2025-02-24:
Based on preceding explanation, should we interpret the code implementations in iommu_two_stage_trans.c and iommu_second_stage_trans.c as indicating that this reference model does not natively support 64K page sizes, thereby defaulting to 4K page translations?

HDntown, 2025-02-26:
Thank you, I get it now.