riscv-non-isa/riscv-iommu #499: global mapping in the IOTINVAL.VMA operands and operations

devotee01, 2025-02-21:
question：
      example：if use command iotinval.vma with  gv=0, av=1,pscv=0 to invalid the all address-translation cache entry in first-stage leaf page table, how to understand function of global mappings description in spec? 
1.  if iova='h1000,and command iotinval.vma ADDR[63:12]=‘h1000, so should invalid the all address-translation cache entry. but in this situation, if the g field of pte is 1 or 0 , iotinval.vma  whether can normally invalid the all address-translation cache entry ？
2. global mapping has what influence for   iotinval.vma  to invald address-translation cache entry?

ved-rivos, 2025-02-21:
An iotinval can be used to invalidate global mappings by setting PV to zero to indicate that the invalidation is not specific to an address space. A pte with g set to 1 is used to designate a global mappings. Global mappings can be used across address spaces.

devotee01, 2025-02-24:
> An iotinval can be used to invalidate global mappings by setting PV to zero to indicate that the invalidation is not specific to an address space. A pte with g set to 1 is used to designate a global mappings. Global mappings can be used across address spaces.

how to understand this global mappings? what is this situation global mappings?

ved-rivos, 2025-02-24:
Global mappings being global can be reused across address  spaces.  This is useful for kernel pages that remain consistent across all processes, reducing TLB capacity overhead and improving performance. These mappings are mainly used for frequently accessed system structures like kernel code and data. 

The virtual memory section of most computer architecture or operating systems books will be useful to read further on this topic. OSDev and Linux kernel documentation are also good sources

closed 2025-02-24 by ved-rivos