riscv-non-isa/riscv-iommu #502: IOFENCE.C completion  will occur interrupt?

devotee01, 2025-02-25:
whether when IOFENCE.C completion ,then  will occur interrupt?

devotee01, 2025-02-25:
> whether when IOFENCE.C completion ,then will occur interrupt?

[image]

ved-rivos, 2025-02-25:
What is your interpretation of the text that you Hugh highlighted ? Will there be an interrupt or not ?

devotee01, 2025-02-26:
> What is your interpretation of the text that you highlighted ? Will there be an interrupt or not ?
my understanding as follows: 
    when use iofence.c command, we can configure the address of seteipnum_le/seteipnum_be register to control the Byte order, and  when iofence.c completion, wiil occur msi interrupt ?

ved-rivos, 2025-02-26:
In the highlighted text what does it say is caused when the address is an imsic register ?

Also was this question previously answered here 
https://github.com/riscv-non-isa/riscv-iommu/issues/473

devotee01, 2025-02-26:
> In the highlighted text what does it say is caused when the address is an imsic register ?
> 
> Also was this question previously answered here [#473](https://github.com/riscv-non-isa/riscv-iommu/issues/473)

[image]
this wsi value whether is optional for  0 or 1, and 0 as wire-singal interrupt  and 1 as wsi interrupt ?

ved-rivos, 2025-02-26:
Before we switch topics, I would like to understand your interpretation of the first highlighted sentence. Was there any ambiguity or misinterpretation when the first sentence said “cause an external interrupt notification on completion of iofence.c” ?

devotee01, 2025-02-26:
> Before we switch topics, I would like to understand your interpretation of the first highlighted sentence. Was there any ambiguity or misinterpretation when the first sentence said “cause an external interrupt notification on completion of iofence.c” ?

for first highlighted sentence, i think when iofence.c completion, iommu will occur a msi interrupt ? for this sentence, I don't really understand it either.

ved-rivos, 2025-02-26:
That’s interesting. So is your confusion really that whether an external interrupt can be an MSI?

devotee01, 2025-02-27:
> That’s interesting. So is your confusion really that whether an external interrupt can be an MSI?
yes, i want to confirm whether an external interrupt can be an MSI?
[image]
this description will occur wire-signal interrupt, and the description of section 5.15 as follows:

[image]
when iofence.c completion only wire-signaled-interrupts set to 1.

but as follow desciption softwore can configure the ADDR[63:2] operand to specify the address of
the seteipnum_le/seteipnum_be to cause an external interrupt notification on IOFENCE.C completion. then the register seteipnum_le/seteipnum_be desciption in IMSIC chapter, this register use to receiving MSI write.  
so， about iofence.c completion,this two point whether has contradiction?

[image]

ved-rivos, 2025-02-27:
External interrupts to a hart may be signaled using a wire or a message . Does that clear up your confusion ?

ved-rivos, 2025-02-27:
And if you configure iofence to cause a wire signaled interrupt and setup the addesss/data to cause a message signaled interrupt then both will happen

devotee01, 2025-02-27:
> And if you configure iofence to cause a wire signaled interrupt and setup the addesss/data to cause a message signaled interrupt then both will happen
your means is ：iofence operand wsi configure as 0，and configure ADDR[63:2],DATA, then  will occur  signaled interrupt and 
message signaled interrupt ?

ved-rivos, 2025-02-27:
Wsi requests wire signaled interrupts 
Addr/data pair can be used to request message signaled interrupt. You can request either of them or both of them on completion of iofence

devotee01, 2025-02-27:
> Wsi requests wire signaled interrupts Addr/data pair can be used to request message signaled interrupt. You can request either of them or both of them on completion of iofence

so，need to configure iofence operand wsi to decide whether need wire signaled interrupts and  message signaled interrupt , if this ,description about fence_w_ip of section 5.15 whether need to modify?

ved-rivos, 2025-02-27:
> so，need to configure iofence operand wsi to decide whether need wire signaled interrupts and message signaled interrupt , if this ,description about fence_w_ip of section 5.15 whether need to modify?

You misinterpreted what I wrote. Let me try again:

When iofence.c completes:
1. If WSI is 1, then a wire signaled interrupt occurs
2. If ADDR/DATA operands are configured to the address of the seteipnum* registers of an IMSIC with a valid interrupt identity, then a message signaled interrupt occurs.
3. If both 1 and 2 are true then both occur i.e. a wire signaled interrupt occurs and a message signaled interrupt occurs.

devotee01, 2025-03-04:
> You misinterpreted what I wrote. Let me try again:
>
> When iofence.c completes:
>
> 1. If WSI is 1, then a wire signaled interrupt occurs
> 2. If ADDR/DATA operands are configured to the address of the [...]

devotee01, 2025-03-05:
in this, 

> You misinterpreted what I wrote. Let me try again:
>
> When iofence.c completes:
>
> 1. If WSI is 1, then a wire signaled interrupt occurs
> 2. If ADDR/DATA operands are configured to the address of the [...]

i want to confirm that for step2 it can‘t need to config WSI? and for this DATA shoud be what ?

ved-rivos, 2025-03-05:
If you want a wire signaled interrupt then set wsi to 1. If you want a msi then configure the address and data pair to generate the msi you want. If you want both a wire signaled interrupt and a message signaled interrupt then set wsi as 1 and the address and data pair to cause the msi you want

devotee01, 2025-03-06:
> If you want a wire signaled interrupt then set wsi to 1. If you want a msi then configure the address and data pair to generate the msi you want. If you want both a wire signaled interrupt and a message signaled interrupt then set wsi as 1 and the address and data pair to cause the msi you want

if i just want to occur msi，then the ADDR[63:2] set to the address of the seteipnum* registers， and DATA should be set to what value?   thanks！

devotee01, 2025-03-06:
> If you want a wire signaled interrupt then set wsi to 1. If you want a msi then configure the address and data pair to generate the msi you want. If you want both a wire signaled interrupt and a message signaled interrupt then set wsi as 1 and the address and data pair to cause the msi you want
 No matter what wire signaled interrupt or msi， them occur Conditions is iofence.c command  complete ?

ved-rivos, 2025-03-06:
When conditions do complete iofence.c occur then it will complete . The type of interrupt requested on completion is not a condition for completion