riscv-non-isa/riscv-iommu #503: how to control interrupt file big/little endian in IMSIC ？

devotee01, 2025-02-26:
[image]

1. whether write  interrupt identity number in little-endian byte order to seteipnum le, then IMSIC accept msi is  little-endian byte order?
2. whether write  interrupt identity number in big-endian byte order to seteipnum be, then IMSIC accept msi is  big-endian byte order?

ved-rivos, 2025-02-26:
I think these sentences are very clear in what they say. How did you interpret them ? What is your answer to those two questions ?

ved-rivos, 2025-02-26:
I will answer your question but I wanted to use this opportunity to understand if there is a better way to write the spec to make it more broadly understood. I am sure you spent a good amount of time studying that text before creating an issue - so there is likely some room for improvement. So please humor me a bit here if you will.

ved-rivos, 2025-02-27:
The data value written to the seteipnum_le is interpreted using little-endian byte order. The data value written to seteipnum_be is interpreted using the big-endian byte order. If the data value written to these registers is a valid interrupt identity, per the respective byte order. then an interrupt is pending else the write is ignored.

closed 2025-02-27 by ved-rivos

devotee01, 2025-02-28:
> The data value written to the seteipnum_le is interpreted using little-endian byte order. The data value written to seteipnum_be is interpreted using the big-endian byte order. If the data value [...]
the operation about interrupt identify number write to seteipnum_be or seteipnum_le register  by iommu or software? under verify situation, weather software need to simulate this process about interrupt identify number write to seteipnum_be or seteipnum_le register ?

ved-rivos, 2025-02-28:
The behavior is the same whether the write was done by software or hardware

devotee01, 2025-02-28:
> The behavior is the same whether the write was done by software or hardware

1. so ,how to get this data, this data actually is  interrupt identity number ? 

2. the message data whether is  interrupt identity number ?

ved-rivos, 2025-02-28:
1. Yea 
2. yes