riscv-non-isa/riscv-iommu #504: the range of the message data when occur a msi interrupt ?

devotee01, 2025-02-27:
[image]

[image]
so, the correctly range of message data  is 0~2047 or the value written might be limited to a
16-bit range, with the upper 16 bits always being zeros ?

ved-rivos, 2025-02-27:
What is your answer ? How did you interpret “is in the range 0  - 2047” ? Did you think encoding that range need upper 16 bits to be non zero ?

devotee01, 2025-02-27:
> What is your answer ? How did you interpret “is in the range 0 - 2047” ? Did you think encoding that range need upper 16 bits to be non zero and how many bits are needed to encode 2047 ?

i  think message data is interrupt identify, so message data range should be in 0~2047. i need to confirm this.

ved-rivos, 2025-02-27:
In PCIe, the MSI address/data pairs are configured in the MSI capability or when MSI-X is used in the MSI-X table. When the 64-bit address capable bit is 1 in the MSI capability, the MSI message address may be 64-bit. Likewise, the width of the data supported by the function is enumerated by the Extended Message Data Capable bit in the MSI capability. When this capability bit is supported, the device supports an extended data field configuration which is used as the value for the bits 31:16 of the MSI message write. In all cases the Memory Write transaction is a DWORD memory write with all 4 byte enables set to 1.

The AIA IMSIC supports interrupt identities that can range from 0 to 2047. A system that supports AIA does not require support in functions for the Extended Message Data Capability for any of the interrupt identities.

closed 2025-02-27 by ved-rivos