riscv-non-isa/riscv-iommu #506: confirm msi_address is 32 bit?

devotee01, 2025-02-27:
[image]

[image]

[image]

[image]

ved-rivos, 2025-02-27:
MSI address need not be restricted to be 32-bit wide. In PCIe, the MSI address is configured in the MSI capability or when MSI-X is used in the MSI-X table. When the 64-bit address capable bit is 1 in the MSI capability, the MSI message address may be 64-bit.

closed 2025-02-27 by ved-rivos

devotee01, 2025-02-28:
> MSI address need not be restricted to be 32-bit wide. In PCIe, the MSI address is configured in the MSI capability or when MSI-X is used in the MSI-X table. When the 64-bit address capable bit is 1 in the MSI capability, the MSI message address may be 64-bit.

i'm sorry to find the 64-bit address capable bit, where is this bit?

ved-rivos, 2025-02-28:
MSI control register - bit 7