riscv-non-isa/riscv-iommu #510: AMOAND operation step

devotee01, 2025-02-28:
[image]

[image]

where is the  AMOAND operation step for  making interrupt file’s eip array into the interrupt-pending
bits of the MRIF?

ved-rivos, 2025-02-28:
As the highlighted text says - if there is a need to clear a pending bit then amoand can be used

closed 2025-02-28 by ved-rivos

devotee01, 2025-02-28:
> As the highlighted text says - if there is a need to clear a pending bit then amoand can be used

for amoand operand, just need to change AMOOR of step4 to AMOAND ?

devotee01, 2025-03-03:
> AMOOR enables you to atomically set one or more bit and AMOAND with a suitable mask enables you to clear one or more bits atomically

where is mask enables?

ved-rivos, 2025-03-03:
There should be a comma in mg statement between the word - mask - and the word - enables  . Sorry. Besides noting this missing punctuation in mg reply, was there a question ?