riscv-non-isa/riscv-iommu #512: ttyp value confirm

devotee01, 2025-02-28:
[image]
1. if ttyp is 3, then whether iotval is 3 ? 
2. when ttyp is 3 and have no guest-page fault ,whether iotval2 is 0?

[image]

how to understand this  implicit memory access ?

ved-rivos, 2025-02-28:
When ttyp is 3, and a fault is reported then the iotval has a value correspond to the cause code of that fault 

The value reported kn iotval2 depends on the cause code and not ttyp 

Implicit memory accesses are memory accesses originated by the iommu itself. Explicit memory accceas are on the other hand memory accesses originated by devices associated with an iommu

closed 2025-02-28 by ved-rivos

devotee01, 2025-03-07:
> When a fault is reported then the iotval has a value correspond to the cause code of that fault and not ttyp
> 
> The value reported kn iotval2 depends on the cause code and not ttyp
> 
> Implicit memory accesses are memory accesses originated by the iommu itself. Explicit memory accceas are on the other hand memory accesses originated by devices associated with an iommu

the TTYP has multi as: 0,1,2,3,4,5,6,7,8,9; spec just descript “TTYP is a transaction with an IOVA” and “TTYP is a PCIe message
request”, for other TTYP, when occur fault , should also report to iotval ? 
2. below description  means is that if this transaction type is IOVA，then when occur fault，report cause code to iotval ?
[image]
3. for value reported in iotval2 depends on the cause code and not ttyp， this not ttyp indicate  TTYP is out 0~9 of range ? about this whether has more detail description ?

devotee01, 2025-03-07:
[image]
when occur fault, cause store cause code ，and  iotval also store cause code ?

devotee01, 2025-03-07:
> [image] when occur fault, cause store cause code ，and iotval also store cause code ?

ved-rivos, 2025-03-07:
1. Except for message transaction type, all other valid transaction types are associated with an iova. 
2. Iotval2 is only applicable for guest page faults

ved-rivos, 2025-03-07:
> when occur fault, cause store cause code ，and iotval also store cause code ?

When a fault occurs, if the transaction type had a IOVA then that iova is reported in iotval. If the fault is a G-stage page fault then the iotval2 reports the GPA along with information about whether the G-stage access was an implicit read or a implicit write.