riscv-non-isa/riscv-iommu #515: msi address need to  aligned to 4-byte ?

devotee01, 2025-03-03:
[image]
as spec description ,this access address whether is msi address that will need  to translate to spa ?

ved-rivos, 2025-03-03:
The address used for a memory write must be 4 byte aligned to cause an MSI. Please state your question

ved-rivos, 2025-03-03:
PS: what’s written in the spec is usually correct . If is not necessary to create an issue just to reconfirm if what is written is correct

devotee01, 2025-03-04:
> The address used for a memory write must be 4 byte aligned to cause an MSI.

this memory write  whether Can be understood as device seed a address translation request?

ved-rivos, 2025-03-04:
The iommu does an address translation when requested by the io bridge or a device. The same holds for this write under discussion as well.

devotee01, 2025-03-05:
> The iommu does an address translation when requested by the io bridge or a device. The same holds for this write under discussion as well.
satisfy
ok,i just want to confirm if this address is a msi, then this address not only satisfy  (A >> 12) & ~msi_addr_mask = (msi_addr_pattern & ~msi_addr_mask),but also A must be aligned to 4-byte ？

devotee01, 2025-03-06:
> The address A must be 4 byte aligned.

why  must make A(guest physical address)  be  4 byte aligned， What factors to take into account ？