riscv-non-isa/riscv-iommu #518: MSI page table entries  is 2^k，what is this k ?

devotee01, 2025-03-05:
[image]

if the value of  DC.msi_addr_mask has what many "1" ,the k is count of 1 ？example：if  DC.msi_addr_mask='b0010_1010_1010, then this k is 5 ?

closed 2025-03-05 by ved-rivos

devotee01, 2025-03-06:
> Yes. Thats right.
thanks for your write back！
how to keep the start of the table is aligned to a 4-KiB page address ？ 
where is a page address? 
what is difference between page address  and  “MRIF Address” of MSI PTE ？

ved-rivos, 2025-03-06:
1. By allocating memory that starts at a 4KiB aligned address.
2. Page Address is the address where the MSI page table is located in memory.
3. A MSI page table entry in MRIF mode holds the address of the MRIF

devotee01, 2025-03-07:
> 1. By allocating memory that starts at a 4KiB aligned address.
> 2. Page Address is the address where the MSI page table is located in memory.
> 3. A MSI page table entry in MRIF mode holds the address of the MRIF
thanks, i want to confirm again.
for step2, page address whether is : m | （I x16) ? spec description as below is 

[image]

for  step3, the address of the MRIF is: MRIF Address[55:9]  ? spec description as below is 

[image]

ved-rivos, 2025-03-07:
Yes

devotee01, 2025-03-10:
> Yes

confirm again， 
1. so this  need to make sure (m |(Ix16)) is aligned to a 4-KiB , when msi page table has 256 or fewer entries?
2.  make sure MRIF Address[55:9] is aligned to 2^k x 16-byte address boundary , when msi page table has 2^k>256 entries?
3. what value is byte address boundary？

ved-rivos, 2025-03-10:
1. “M” must be aligned to 4K
2. yes 
3. Only bits 55:9 can be specified and the bits 8:0 are always 0. That implies that the MRIF must be aligned to a 512 byte address boundary . If it is aligned to 4K or any higher power of 2 then it is also aligned to 512 byte

devotee01, 2025-03-11:
> 1. “M” must be aligned to 4K
> 2. yes, the base address determines by 55:9 with the bits 8:0 set to 0
> 3. Only bits 55:9 can be specified and the bits 8:0 are always 0. That implies that the MRIF must be aligned to a 512 byte address boundary . If it is aligned to 4K or any higher power of 2 then it is also aligned to 512 byte

for question 2, make sure MRIF Address[55:9] is aligned to 2^k x 16-byte address boundary, what is this "byte address boundary "

ved-rivos, 2025-03-11:
A address boundary aligned to (2^k x 16 ) bytes

devotee01, 2025-03-20:
> A address boundary aligned to (2^k x 16 ) bytes
ok, I mistook "-" of "2^k x 16-byte address boundary" as subtraction. 

when msi page table has 256 or fewer entries,  spec says: "the start of the table is aligned to a 4-KiB
page address in real physical memory“ 
this means just keep "m" is  aligned to a 4-KiB ?