riscv-non-isa/riscv-iommu #519: each guest interrupt file occupy 4k  space size？

devotee01, 2025-03-06:
[image]

example: for RISC-V 64 single hart, there is  guest interrupt file count is 64个，each  guest interrupt file have max-2047 "interrupt identifies" , i want to know how "interrupt identifies" occupy what space ?

ved-rivos, 2025-03-06:
In each interrupt file there is a bit per interrupt identify to hold pending interrupts

ved-rivos, 2025-03-06:
__

closed 2025-03-06 by ved-rivos

devotee01, 2025-03-11:
> In each interrupt file there is a bit per interrupt identify- the external interrupt pending bit array - to hold pending interrupts

1. as your description, the  interrupt identify is one bit in "interrupt-pending" field of one interrupt file ? 
2. i want to know, for one interrupt file, how rv_iommu to handle when iommu occur interrupt quantity more than 2048 ?

devotee01, 2025-03-11:
> 1. as your description, the  interrupt identify is one bit in "interrupt-pending" field of one interrupt file ?
> 2. i want to know, for one interrupt file, how rv_iommu to handle when iommu occur [...]

devotee01, 2025-03-11:
> In each interrupt file there is a bit per interrupt identify- the external interrupt pending bit array - to hold pending interrupts

devotee01, 2025-03-11:
> > In each interrupt file there is a bit per interrupt identify- the external interrupt pending bit array - to hold pending interrupts

ved-rivos, 2025-03-11:
1. There is one pending bit per interrupt identify 
2. An IMSIC has a limit of 2K identities

devotee01, 2025-03-12:
> 1. There is one pending bit per interrupt identify
> 2. An IMSIC has a limit of 2K identities

why can‘t upload the picture now?
as spec description,  interrupt identify is 1~2047, why is limited to 2k ?

ved-rivos, 2025-03-12:
2K identifies per hart were deemed sufficient by the AIA architecture .

devotee01, 2025-03-13:
> 2K identifies per hart were deemed sufficient by the AIA architecture .

for num of 2047 interrupt identify, i think there is 2047bit/8=255byte ?  how to compute the 2k?

ved-rivos, 2025-03-13:
Please clarify which 2K you are referring to

devotee01, 2025-03-16:
> Please clarify which 2K you are referring to
 I referring to 2k is your reply when 5 days ago，as follow : “An IMSIC has a limit of 2K identities”

ved-rivos, 2025-03-16:
2047/8 is 255 bytes and 7 bits. The bit 0 is always 0 since the identity 0 is not valid.

devotee01, 2025-03-16:
> 2047/8 is 255 bytes and 7 bits. The bit 0 is always 0 since the identity 0 is not valid.

1. so, for one interrupt file, if External interrupt more than 2048 num, then the interrupt file have no interrupt identify，then how rv_iommu to handle these interrupt ? 
2. Whether the iommu generates an exception？

ved-rivos, 2025-03-16:
Reading the AIA specification - that you are pasting from - will reveal the answer to these questions about handling of an MSI write when the interrupt identity is not supported - which includes the case where the identity is 0 or >2047. I quote a few sections below to help direct you to the sections you should read.

1. Section "Memory region for an interrupt file":
> A write to seteipnum_le is ignored if the value written is not an implemented interrupt identity number in little-endian byte order.

> A write to seteipnum_be is ignored if the value written is not an implemented interrupt identity number in
big-endian byte order.

2. Section "MSI PTE, MRIF mode":

> If either A[11:3] or D[31:11] is not zero, or if bit 2 of A is one and big-endian MSIs are not supported, then the incoming write is accepted but discarded.

Please ask if you need help with reading any other part of the specification.

devotee01, 2025-03-16:
> Reading the AIA specification - that you are pasting from - will reveal the answer to these questions about handling of an MSI write when the interrupt identity is not supported - which includes the [...]

my question is : for one interrupt file, if External interrupt more than 2048 num, then the interrupt file have no sufficient interrupt identify, then how rv_iommu to handle these interrupt ? and Whether the iommu generates an exception？

for you say "Section "MSI PTE, MRIF mode" ", 
1.  my question：A is MSI ADDRESS?
2.  what situation “the write data is D when interpreted in the byte order indicated by bit 2 of A” ？

ved-rivos, 2025-03-16:
All your questions have been answered a few times over in a few different ways . I am at a loss of words to explain this any further. Sorry

ved-rivos, 2025-03-16:
I will try one more time: 
The specifications are written to state what happens when a condition occurs . If the iommu specification does not say an exception occurs then it really means that. When the AIA specification says that writes are ignored or discarded, it really means that. If the AIA specification and iommu specification do not say that an exception occurs., they really mean that. 

The meaning of A is provided in the section I provided a reference to - please read that

The section I referenced also shows a picture of how the seteipnum_le and seteipnum_be registers are laid out. The little endian register is at assess 0 and big endian register is at address 4. The bit 2 of an address being 1 and bits 11:3 and bits 1:0 being 0 means the write is to the big endian register at offset 4 of the page and when big endian register is written the data is interpreted as big endian

devotee01, 2025-03-16:
> I will try one more time: The specifications are written to state what happens when a condition occurs . If the iommu specification does not say an exception occurs then it really means that. When the AIA specification says that writes are ignored or discarded, it really means that. If the AIA specification and iommu specification do not say that an exception occurs., they really mean that.
> 
> The meaning of A is provided in the section I provided a reference to - please read that
> 
> The section I referenced also shows a picture of how the seteipnum_le and seteipnum_be registers are laid out. The little endian register is at offset 0 and big endian register is at offset 4. The bit 2 of an address being 1 and bits 11:3 and bits 1:0 being 0 means the write is to the big endian register at offset 4 of the page and when big endian register is written the data is interpreted as big endian

okey,thanks ! I will try to restart to understanding refer to above section. thanks

devotee01, 2025-03-16:
> You are welcome. Please don’t hesitate to ask if there are further questions

ok,thanks！