riscv-non-isa/riscv-iommu #523: what distinction between supervisor mode and  hypervisor extend ?

devotee01, 2025-03-11:


devotee01, 2025-03-11:
> No description provided.

[image]

devotee01, 2025-03-11:
> > No description provided.
> 
> [image]

devotee01, 2025-03-11:
[image]

ved-rivos, 2025-03-11:
Supervisor mode is a privilege mode used to execute privileged software such as operating systems and hypervisors. A hypervisor enables multiple virtual machines to run on a physical machine . 

For more details please read about the riscv privileged architecture and the privileged ISA manual