riscv-non-isa/riscv-iommu #528: Why does msi address need to be translated by fsc page table?

LuXuKun, 2025-03-16:
Now that there is already msi remapping page table, why does msi address need to be translated by fsc page table too? Does it require self-developed user mode driver(not virtualization machine scenario) not only to perform DMA mapping on normal memory, but also to perform DMA mapping on MSI addresses? Will this lead to wastage of performance and page table memory?

ved-rivos, 2025-03-16:
If a memory write is done using a virtual address then in order to determine that an address is an MSI address, it needs to be first translated to a guest physical address by the first stage page table. Without doing that its not possible to determine its a MSI address. When the memory write is done using a guest physical address then the translation using first stage page table is not needed.

LuXuKun, 2025-03-16:
It seems not so friendly to IOMMU interrupt remapping driver? When the IOMMU irq domain allocates a new msi irq for pcie device, it needs to map the msi address in fsc page table first. Besides, different pcie devices may have difference fsc page tables, which brings more complexity for IOMMU irq domain. Will it be more convenient if translation of MSI and DMA addresses can be totally seperated?

ved-rivos, 2025-03-16:
If the PCIe device is configured to not generate an MSI write with PASID then the IOVA of that MSI write a GPA and there is no need to translate using the first stage page table.

LuXuKun, 2025-03-16:
Sorry, I am still confused. Does it mean a user mode driver (not an VM) which needs both dma and msi has to use iohatp for dma remapping and left iosatp bare?

ved-rivos, 2025-03-16:
If the user mode driver in this example uses VA for DMA then it would configure the DMA accesses then it would have a non-bare `iosatp`. It may have a bare iohgatp. Both DMA write and MSI are nothing but MSI writes - a MWr transaction. In this case it is expected that the Mwr for MSI is done without a PASID and the other DMA are done with a PASID. Since MSI writes do not have a PASID they will not need address transaction through a first stage page table - their address is just a GPA. Since `iohgatp` is bare, the GPA is just a PA. 

Two stage address translation is always active in the IOMMU - however, either stages can be effectively disabled by setting the page table of that stage to Bare.

LuXuKun, 2025-03-16:
That makes more sense. Thanks for answering my question.

I still have some questions due to little knowledge of hardware microarchitecture implementation.

Is MSI writes without PASID equal to translated request as is stated in IOMMU Spec 3.3 "Process to translate an IOVA"? How does pcie distinguish the dma write and msi write? In other words, how does pcie issue msi writes with PASID and issue dma writes PASID? Is it configured by software or automatically done by hardware?

ved-rivos, 2025-03-16:
Devices need to be configured by SW with both a) the address to use for a memory write b) the PASID to use. For MSI, the address and data values to use for the write are configured in either the MSI table or the MSI-X table. There is not architectural way to associate a PASID with MSI. When a device generates an MSI using the address/data pair configured in MSI/MSI-X tables it would generate those memory write without PASID.

LuXuKun, 2025-03-16:
Great! I understand now. Thanks a lot. Best regards.

ved-rivos, 2025-03-16:
Glad to be able to help!

closed 2025-03-16 by ved-rivos