riscv-non-isa/riscv-iommu #529: when MSI page table entries >256 entries, Whether msi mode must be mrif mode ?

devotee01, 2025-03-20:
I have a question that for the start of the table is aligned to what . 
1.  if MSI page table entries >256 entries, and in basic mode.  how to keep start of the table is aligned 2^k * 16 ？ 
2. in other word, the num of MSI page table entries whether Relate to the MSI PTE mode?

devotee01, 2025-03-20:
> I have a question that for the start of the table is aligned to what .
> 
> 1. if MSI page table entries >256 entries, and in basic mode.  how to keep start of the table is aligned 2^k * 16 ？
> 2. in other word, the num of MSI page table entries whether Relate to the MSI PTE mode?

[image]

ved-rivos, 2025-03-20:
The number of entries in an MSI page table is 2^k where k is the number of bits that are ones in the MSI address mask used to extract the interrupt file number. Each entry in the MSI page table is 16 bytes wide. Thus when the number of entries is 256 i.e. k is 8, the size of the table is 2^8 * 16 = 4096. If the number of entries is fewer than 256, the table is always required to be aligned to a 4 KiB address boundary. When the number of entries is > 256 the table is required to be aligned to a 2^k * 16 byte address boundary. For instance, if k is 9 then number of entries is 512 and the table size is 8 KiB and the table is required to be aligned to a 8 KiB address boundary and if k is 10 to a 16 KiB address boundary and so on.

When a MSI PTE is in MRIF mode, the PTE holds the address of a MRIF. Each MRIF is required to be aligned to a 512-byte address boundary.

closed 2025-03-20 by ved-rivos

devotee01, 2025-03-21:
> The page table base address - m - must be aligned as required. This is irrespective of whether some entries in the page table are of basic mode and others are of mrif mode

so, for basic mode , if table size is 8KiB, which address should be aligned to 8KiB ? 
in another word, for basic mode , if table size is 8KiB, set which address to keep aligned to  8KiB ?

ved-rivos, 2025-03-21:
The alignment in question here is that if the table itself. The page table itself must be aligned

devotee01, 2025-03-23:
> The alignment in question here is that if the table itself. The page table itself must be aligned . Please do not conflate the alignment requirement of the page table with the configuration of specific entries in the page table

ok, for basic mode , if table size is 8KiB, how to keep page table itself must be aligned to 8Kib ?

ved-rivos, 2025-03-23:
What is that you call “basic mode”?

ved-rivos, 2025-03-23:
A address is aligned to 8K if the bits 12:0 of the address is 0. Here in the process to translate an MSI, the address of the page table is in the variable “m”. By allocating the memory for the MSI page table starting at an address where bits 12:0 are all 0 you can ensure that the said page table is aligned to 8K

devotee01, 2025-03-25:
> A address is aligned to 8K if the bits 12:0 of the address are 0. Here in the process to translate an MSI, the address of the page table is in the variable “m”. By allocating the memory for the MSI page table starting at an address where bits 12:0 are all 0 you can ensure that the said page table is aligned to 8K

So according your idea, msi pte is basic translate mode or mrif mode , if table size is 16KiB（k=10），then  variable “m” 13:0 should be 0 ？

ved-rivos, 2025-03-25:
It’s not my idea . It’s how the specification is written . In your example If k=10 then the page table must be aligned to 16K address boundary. There are 1024 PTEs in such a table - and an entry may be invalid or be in basic translate mode or in MRIF mode .

devotee01, 2025-03-25:
> It’s not my idea . It’s how the specification is written . In your example If k=10 then the page table must be aligned to 16K address boundary. There are 1024 PTEs in such a table - and an entry may be invalid or be in basic translate mode or in MRIF mode .

1. why "and an entry may be invalid or be in basic translate mode or in MRIF mode "? 
2. page table must be aligned, Does hardware need to be aligned or software needs to be aligned ? I'm not sure which parameter to set or configure to achieve alignment

devotee01, 2025-03-25:
> Bit 0 of the first doubleword of an MSI PTE is field V (Valid). When V = 0, the PTE is invalid. If V = 1 and the custom-use bit C = 0, then bits 2:1 of the first doubleword contain field M (Mode). If M = 3, the MSI PTE specifies basic translate mode for accesses to the page, and if M = 1, it specifies MRIF mode.
> 
> The step 6 of the MSI address translation process states: "Let m be (DC.msiptp.PPN x 212).". Hope that helps you understand what to configure to get aligned "m" value.
The ppn value needs to be configured according to the different sizes of the page table size alignment. For example, if k=11, then the size of the table is 32k, and bit[14:0] needs to be 0, so PPN[1:0]=0, m=PPN<<12 will achieve the purpose of page table alignment to 32k。 
Above is my understanding, you see correct ？thanks!

ved-rivos, 2025-03-25:
Yes. I am glad you finally got it.

devotee01, 2025-03-26:
> Yes. I am glad you finally got it.
thanks for you give me Constant explanation. I appreciate it!