riscv-non-isa/riscv-iommu #534: about understand of "global mappings" when use IOTINVAL.VMA cmd to invalidate first page table？

devotee01, 2025-03-21:
example: 
if use IOTINVAL.VMA,GV=1,AV=1,PSCV=0, to invalidate first stage page table; under this configuration, as spec description: "Invalidates all address-translation cache entries that contain first-stage leaf page table entries, including those that contain global mappings, corresponding to the IOVA in ADDR operand, for all VM address spaces associated with the GSCID operand " 
follow  is my understand：
1. Can it be understood as that Whether or not the value of AV matches that of ADDR, or Whether or not the value of GV matches that of GSCID, As long as "g" of PTE field is 1, this PTE will be invalidated ？ 
2.  why as long as PSCV=0, invalidate operation  will contain "global mappings" , Do these two have any special meaning in design ?

ved-rivos, 2025-03-21:
1. The quoted text states that the invalidation of entries that match the GSCID occurs regardless of the global (g) attribute of those entries . This is almost opposite of your takeaway 
2. when PSCV is 1, it indicates that the command operates on a specific process address space identified by the PSCID . When it is 0, there is no process address space identified and the command operates globally across all applicable process address spaces

closed 2025-03-21 by ved-rivos

devotee01, 2025-03-23:
> 1. The quoted text states that the invalidation of entries that match the GSCID occurs regardless of the global (g) attribute of those entries . This is almost opposite of your takeaway
> 2. when PSCV [...]

1. Can you be a little more specific? I'm a little confused about this ?

ved-rivos, 2025-03-23:
Are you familiar with what global mapping are ?