riscv-non-isa/riscv-iommu #537: msi data and address limit understand

devotee01, 2025-03-26:
as spec description: "Established standards (in particular, for PCI and PCI Express) dictate that an individual messagesignaled interrupt (MSI) from a device takes the form of a naturally aligned 32-bit write by the device, with the address and value both configured at the device (or device controller) by software. Depending on the versions of the standards to which a device or controller conforms, the address might be restricted to the lower 4-GiB (32-bit) range, and the value written might be limited to a 16-bit range, with the upper 16 bits always being zeros."
above about addr and data limit, these limit is only just for pcie device?

ved-rivos, 2025-03-26:
Depends on standard and version of the standard to which the device complies . For pcie, the decide is 64 bit address capable if it sets the 64 bit capable bit in the MSI capability

closed 2025-03-26 by ved-rivos

devotee01, 2025-03-26:
> Depends on standard and version of the standard to which the device complies . For pcie, the decide is 64 bit address capable if it sets the 64 bit capable bit in the MSI capability

Whatever the standard and version of the standard is, the value written might be limited to a 16-bit range, with the upper 16 bits always being zeros?

ved-rivos, 2025-03-26:
Depending on standard and version the data value may be limited to 16 bits. For pcie for example if the device is extended message data capable then it can be programmed to generate a 32 bit data value