riscv-non-isa/riscv-iommu #538: IOFENCE.C have timed out

devotee01, 2025-03-26:
as spec description "If one or more ATS invalidation commands preceding the IOFENCE.C have timed out, then software may make the CQ operational again and resubmit the invalidation commands that may have timed out" 
question: when ATS invalidation commands preceding the IOFENCE.C have timed out, how software  may make the CQ operational again and resubmit  ?

ved-rivos, 2025-03-26:
Software may need to reset the functions that lead to the invalidation timeout. The command queue can be made operational again by clearing the cmd_to error bit .

closed 2025-03-26 by ved-rivos

devotee01, 2025-03-27:
> Software may need to reset the functions that lead to the invalidation timeout. The command queue can be made operational again by clearing the cmd_to error bit .

example : if the command  queue have 10  ATS invalidation commands and one FENCE.C command,  A timeout occurred while executing the fifth command; 
my question: 
1. in this case, the remaining 5 commands will not be executed until the corresponding cmd_to bit is cleared ？ 
2. if Never deal with the cmd_to bit, fence.c command will be executed completely ?
3. if clear  cmd_to bit, The remaining five commands are automatically executed by iommu ?

ved-rivos, 2025-03-27:
When IOFENCE times out, it is not possible to say which of the 10 commands timed out. Software has to assume all of them have timed out.

devotee01, 2025-03-27:
> When IOFENCE times out, it is not possible to say which of the 10 commands timed out. Software has to assume all of them have timed out.
if clear cmd_to bit,  Whether the commands in queue continue to be executed ?

ved-rivos, 2025-03-27:
Yes

devotee01, 2025-03-27:
>fence.c command whether executed completely before clear the cmd_to bit ?

ved-rivos, 2025-03-27:
Please see the spec about command timeout. Since iofence times out it does not complete and since it does not complete the command queue head does not move past iofence