riscv-non-isa/riscv-iommu #539: Fault queue head Read and write attribute validation

devotee01, 2025-03-26:
as spec  description: fqh is WARL, cqh is OR, pqh is WARL ？

ved-rivos, 2025-03-26:
What does the spec say ? Are they writable ?

closed 2025-03-26 by ved-rivos

ved-rivos, 2025-03-26:
From now on please using this single issue to ask all questions instead of opening new issue

devotee01, 2025-04-11:
as issues 546， what is IUL，what does it represent?

ved-rivos, 2025-04-11:
In C, suffixes on integer constants like U, L, or UL explicitly define the constant’s type, such as unsigned, long, or unsigned long. This is important for ensuring correct behavior in expressions, especially with bit shifts or type-specific operations, and for avoiding warnings or unintended type promotions. For example, writing 1UL ensures the constant 1 is treated as an unsigned long, which matters when shifting left by 32 bits on a 64-bit system, or when interacting with variables of that type.

You can find more details about internet and floating point suffixes in any C programming language book. 

Hope that helps

devotee01, 2025-04-16:
Hi,I have a new question:
1.  when address is translating by iommu , whether can write ATS.command to command queue? 
2. If so, what would iommu do in such cases

ved-rivos, 2025-04-17:
The iommu may process and address translation request and an ats command simultaneously. The behavior for both is as specified - just they will occur concurrently

devotee01, 2025-04-17:
> The iommu may process and address translation request and an ats command simultaneously. The behavior for both is as specified - just they will occur concurrently

thanks!

devotee01, 2025-04-17:
I have a new question, for IOFENCE command ,whether there is a situation where the IOFENCE instruction times out ？

devotee01, 2025-04-18:
> > I have a new question, for IOFENCE command ,whether there is a situation where the IOFENCE instruction times out ？
> 
> Please see section 4.1.2: If the `IOFENCE.C` times out waiting on completion of previous commands that are specified to have a timeout, then the `cmd_to` bit in `cqcsr` Section 6.15 is set to signal this condition. The `cqh` holds the index of the `IOFENCE.C` that timed out and all previous commands that are not specified to have a timeout have been completed and committed. In this version of the specification, only the `ATS.INVAL` command is specified to have a timeout.
> 
> A timeout for `ATS.INVAL` occurs if the Function which the Invalidation Request message was sent by this commend does respond with a valid Invalidation Completion message within the time limit specified by the PCIe specficiation.

so, IOFENCE.C itself can't generate timeout; when ATS cmd generate timeout, will generate cmd_to interrupt, and cqh will holds this IOFENCE.C location ?

ved-rivos, 2025-04-18:
It is only an iofence.c that Dan lead to a timeout due to waiting for commands older than if to complete

devotee01, 2025-04-18:
thanks. 
one new question about command queue:
1. Whether it supports writing various commands to the command queue at one time? for example，once  write command : ATS.INVAL ,ATS.PRGR, IODIR.INVAL_DDT, IODIR.INVAL_PDT,IOTINVAL.VMA, IOTINVAL.GVMA to commnad queue, How will iommu handle it? Will these commands be executed in the order they are written?

devotee01, 2025-04-22:
> Here is what the specification states
> 
> > The IOMMU fetches commands from the CQ in order but the IOMMU may execute the fetched commands out of order. The IOMMU advancing cqh is not a guarantee that the commands fetched by the IOMMU have been executed or committed.

so, we can't know the iommu  execute the fetched commands In what order?

devotee01, 2025-04-22:
one question:
     for msi address translation, if the fsc.mode/iosatp.mode as not bare, and the iohgatp.mode is bare. In this situation, assuming the address meets the msi address translation, will iommu perform address translation on msi according to the msi address processing flow?

devotee01, 2025-04-24:
> one question: for msi address translation, if the fsc.mode/iosatp.mode as not bare, and the iohgatp.mode is bare. In this situation, assuming the address meets the msi address translation, will iommu perform address translation on msi according to the msi address processing flow?

> one question: for msi address translation, if the fsc.mode/iosatp.mode as not bare, and the iohgatp.mode is bare. In this situation, assuming the address meets the msi address translation, will iommu perform address translation on msi according to the msi address processing flow?

> so, we can't know the iommu execute the fetched commands In what order?

hi ,if you have time to resolve my question , I would be most grateful.

devotee01, 2025-04-24:
new question：
  as AIA spec description:  "The smallest granularity of address translation implemented by all
IOMMUs is not larger than a 4-KiB page, matching that of standard RISC-V address translation page tables."
 1. my understand : about this  not larger than a 4-KiB page is that page table entries(PTEs)  not larger than a 4-KiB. so for msi  address translation, first stage translation mode(sv32,sv39,sv48,sv57) all supported for msi address translation ?

ved-rivos, 2025-04-24:
> for msi address translation, if the fsc.mode/iosatp.mode as not bare, and the iohgatp.mode is bare. In this situation, assuming the address meets the msi address translation, will iommu perform address translation on msi according to the msi address processing flow?

Yes, the IOMMU will perform MSI address translation. Please see step 18 of the process to translate an IOVA (section 3.3). The step 18 does not depend on the `iohgatp.mode`.

ved-rivos, 2025-04-24:
> as AIA spec description: "The smallest granularity of address translation implemented by all IOMMUs is not larger than a 4-KiB page, matching that of standard RISC-V address translation page tables."

That statement says that the smallest granularity of address translation is 4 KiB i.e. the minimum page size that can be represented in a RISC-V page table is 4 KiB.

> bout this not larger than a 4-KiB page is that page table entries(PTEs) not larger than a 4-KiB. so for msi address translation, first stage translation mode(sv32,sv39,sv48,sv57) all supported for msi address translation ?

Correct. All of these modes support mapping a 4 KiB page.

devotee01, 2025-04-25:
> That statement says that the smallest granularity of address translation is 4 KiB i.e. the minimum page size that can be represented in a RISC-V page table is 4 KiB.
>
>
> Correct. All of these modes [...]

Thanks!

devotee01, 2025-04-25:
about msi address translation new question:
         1. if first stage page table(iosatp.mode/fsc.mode as bare), the device request address whether is GPA? 
         2. Does the following situation exist: 
                  the request address is gva, then  is translated to GPA by first stage page table. at same time, this GPA satisfied Relational formula  (A >> 12) & ~msi_addr_mask = (msi_addr_pattern & ~msi_addr_mask). at this time iommu, does iommu consider this address request to be an msi address ? and then translated this address by msi address translation process ?

ved-rivos, 2025-04-25:
1. Yes
2. yes. Yes

jones-drew, 2025-04-25:
> Yes, the IOMMU will perform MSI address translation. Please see step 18 of the process to translate an IOVA (section 3.3). The step 18 does not depend on the `iohgatp.mode`.

One thing to keep in mind is that the stage1 page table will need to include an IOVA to GPA mapping for the GPA that will be translated by MSI address translation. In cases where it's expected that the MSI address translation is the only necessary translation for MSI addresses, then the stage1 page table will need identity mappings to avoid faults.

jones-drew, 2025-04-25:
> One thing to keep in mind is that the stage1 page table will need to include an IOVA to GPA mapping for the GPA that will be translated by MSI address translation. In cases where it's expected that [...]

And this is because step 18 strictly follows step 17 and step 17 executes the stage1 translation.

devotee01, 2025-05-07:
> And this is because step 18 strictly follows step 17 and step 17 executes the stage1 translation.

thanks!

devotee01, 2025-05-08:
I have a new  question as follows:

[image]

   1.  what is difference about cause 270,271,273?
   2. how to do or which situation will occur  cause 270?
   3. how to do or which situation will occur  cause 271?
   4.  how to do or which situation will occur  cause 273?

devotee01, 2025-05-12:
> I have a new question as follows:
> 
> [image]
> 
> 1. what is difference about cause 270,271,273?
> 2. how to do or which situation will occur  cause 270?
> 3. how to do or which situation will occur  cause 271?
> 4. how to do or which situation will occur  cause 273?

hi, can you  help me solve my question ,when you have extra time. Thanks!

ved-rivos, 2025-05-12:
> what is difference about cause 270,271,273?

Systems with RAS usually employ mechanisms such as error correcting codes (ECC) to detect data corruption in memory. On a load from memory the ECC on the loaded data is computed and compared against the ECC previously stored with the data. Depending on the code used one or more bits of error may be correctable. If more than the number of bits that can be corrected are corrupted then a data corruption is signaled to the requester of the data. 

The cause 270 occurs when the MSI PTE load detects data corruption. 

The cause 271 occurs when a load from the MRIF - determined by reading the MSI PTE - detects data corruption. 

The cause 273 occurs when an attempt to cause an MSI - a memory write to signal an interrupt - detects and access fault due to violating a physical memory attribute or physical memory protection constraint.

devotee01, 2025-05-13:
> Systems with RAS usually employ mechanisms such as error correcting codes (ECC) to detect data corruption in memory. On a load from memory the ECC on the loaded data is computed and compared against [...]
ok , then for cause 270,271,how to make the data corruption? Will attempting to change the original value of msi pte cause data corruption?

ved-rivos, 2025-05-13:
To create a data corruption you may need to corrupt the ECC. Please check if your system provides hardware error injection capabilities or if you are using a bus transactor then you may be able to signal data corruption to the IP using the protocol used by the bus to signal it