riscv-non-isa/riscv-iommu #542: in MSI address translation procedure. how to make A as GPA?

devotee01, 2025-04-08:
question: how to make A as GPA?  
my understand as follow: 
GPA can only be obtained by converting iova to a first-level address, and there is no other way for iommu to think that this address is the GPA. Am I right ? thanks！

closed 2025-04-08 by ved-rivos

devotee01, 2025-04-08:
> question: how to make A as GPA? my understand as follow: GPA can only be obtained by converting iova to a first-level address, and there is no other way for iommu to think that this address is the GPA. Am I right ? thanks！

question2: How does iommu know if the conversion address is gpa or gva ?

ved-rivos, 2025-04-08:
Please ask the question in the issue #539

ved-rivos, 2025-04-08:
@devotee01 please do not create new issues . I have created #539 for you to ask questions - please ask these two questions in #539