riscv-non-isa/riscv-iommu #555: MSI FLAT MODE

1252005989, 2025-05-15:
Hi,
I want to know whether the scenario where iohgatp.mode=bare, iosatp.mode=bare, and msiptp.mode=flat is reasonable, and whether hardware design needs to handle such a scenario.  For example , in this scenario , hardware design still needs to compare IOVA with msi_addr_mask+msi_addr_pattern and determine whether it points to a MSI PTE

ved-rivos, 2025-05-15:
The hardware should support this configuration. In this case the GPA = IOVA as iosatp.mode is Bare. This GPA, which is same as IOVA, will be used to determine if the GPA is that of an MSI and if so the MSI page tables are used else the G-stage page table determined by iohgatp is used. If the G-stage page table is used then in this example the SPA = GPA as iohgatp.mode is Bare.

1252005989, 2025-05-16:
Thank you for the explanation.

closed 2025-05-16 by 1252005989