riscv-non-isa/riscv-iommu #581: iotval2[1]   need amo process has error  happened ;

asdvvcc123, 2025-07-11:
hi ,sir : 

i think here need amo process has error ,then can use SADE ; 

[image]

thank;

ved-rivos, 2025-07-11:
The guest page faults here are only caused due to G-stage address translation for VS-stage PTEs. Hence they are always implicit accesses. The implicit accesses are implicitly stores if hardware A/D updating is enabled for VS-stage i.e. SADE is 1 else they are implicit loads.

This follows from the following rule:

> When two-stage address translation is active, updates to the D bit in G-stage PTEs may be performed by an implicit access to a VS-stage PTE, if the G-stage PTE provides write permission, before any speculative access to the VS-stage PTE.

ved-rivos, 2025-07-11:
Yes

closed 2025-07-11 by asdvvcc123