riscv-non-isa/riscv-iommu #587: Access Fault during G-Stage walk for PDT walk.

zhuotianshu, 2025-07-17:
When access fault (caused by PMP, PMA, Bus error, etc.) occurs during G-Stage walk, now spec says that:

> If faults occur during second-stage address translation of a then stop and report the fault detected by the second-stage address translation process.

which will report an access fault corresponding to the original access type. However, I wonder why it doesn't report a “PDT entry load access fault" instead, since the fault is actually triggered by accessing the PDT load, and the PDT load is already decoupled from the original access, as their faults are handled separately.

ved-rivos, 2025-07-17:
It's reported as a PDT entry load access fault. A clarification update (https://github.com/riscv-non-isa/riscv-iommu/pull/569/commits/8796a2febb3452bdc2897e46f789fc4066501a7b) to the spec is queued for this.

zhuotianshu, 2025-07-18:
OK, thank you very much.

closed 2025-07-18 by zhuotianshu