riscv-non-isa/riscv-iommu #589: Could DMA request X permission alone?

zhuotianshu, 2025-07-17:
In reference model, a translation request use no_write=1 to request R permisson, and use no_write=0 to request R and W permission. Does it mean that there is no case that DMA does't request R (which only request X, like instruction fetch in CPU)?

ved-rivos, 2025-07-17:
An ATS Translation Request is a PCIe concept and in PCIe there is not concept of requesting execute only permission. In a Translation Response the Exe bit may only be set if the R bit is also set.

zhuotianshu, 2025-07-17:
Thanks. Now I see that no_write is only for ATS translation request. However, even for normal translation request, the reference model still requires read for non-writeAMO requests, regardless of `exec_req`. Is it impossible to request X only?

ved-rivos, 2025-07-17:
> However, even for normal translation request, the reference model still requires read for non-writeAMO requests, regardless of exec_req. Is it possible to request X only?

Could you point to the case.

zhuotianshu, 2025-07-17:
`get_attribs_from_req(
    hb_to_iommu_req_t *req, uint8_t *read, uint8_t *write, uint8_t *exec, uint8_t *priv) {

    *read = ( req->tr.read_writeAMO == READ ) ? 1 : 0;
    *write = ( req->tr.read_writeAMO == WRITE ) ?  1 : 0;

    // The No Write flag, when Set, indicates that the Function is requesting read-only
    // access for this translation.
    // The TA (IOMMU) may ignore the No Write Flag, however, if the TA responds with a
    // translation marked as read-only then the Function must not issue Memory Write
    // transactions using that translation. In this case, the Function may issue another
    // translation request with the No Write flag Clear, which may result in a new
    // translation completion with or without the W (Write) bit Set.
    // Upon receiving a Translation Request with the NW flag Clear, TAs are permitted to
    // mark the associated pages dirty. Functions MUST not issue such Requests
    // unless they have been given explicit write permission.
    // Note ATS Translation requests are read - so read_writeAMO is READ for these requests
    *write = ( (req->tr.at == ADDR_TYPE_PCIE_ATS_TRANSLATION_REQUEST) &&
                 (req->no_write == 0) ) ? 1 : *write;

    // If a Translation Request has a PASID, the Untranslated Address Field is an address
    // within the process address space indicated by the PASID field.
    // If a Translation Request has a PASID with either the Privileged Mode Requested
    // or Execute Requested bit Set, these may be used in constructing the Translation
    // Completion Data Entry.  The PASID Extended Capability indicates whether a Function
    // supports and is enabled to send and receive TLPs with the PASID.
    *exec = ( (*read && req->exec_req &&
                (req->tr.at == ADDR_TYPE_UNTRANSLATED || req->pid_valid)) ) ? 1 : 0;
    *priv = ( req->pid_valid && req->priv_req ) ? S_MODE : U_MODE;
    return;
}`

In the above code, `read` will be one as long as read_writeAMO != WRITE. Is it possible to get attrbute that x=1, r=w=0 from requests?