riscv-non-isa/riscv-iommu #590: MSI translation flow

Chen-rui55, 2025-07-17:
hi ,
I have a question about the MSI translation flow.
In Section 3.3, when only stage-1 translation is enabled, does an MSI access still need to proceed to step 18 to fetch the MSI PTE ( the process defined in Section 3.3.3)  after stage-1 translation? or ,  the GPA is the final MSI address?

ved-rivos, 2025-07-17:
For an IOMMU, two-stage address translation is always active. However, either stage may be effectively disabled by setting the mode for that stage to be Bare. The steps are outlined in section 3.3. First the process invokes step 17 to determine the GPA. When the second-stage is Bare, the GPA is same as the SPA. After step 17, then the step 18 is invoked to determine if MSI address translations are enabled and if so the process in section 3.3.3 is invoked.

ved-rivos, 2025-07-19:
> So my question is: Would there ever be a scenario in software usage where only stage1 is enabled, stage2 is in bare mode, while DC.msiptp.MODE != Off?"

A bare metal OS could use this mechanism to redirect MSIs from one core to another . 

I am not sure Whether it’s a prevalent software use case, but the mechanism is supported by the iommu specification

Chen-rui55, 2025-07-21:
[image]
Sorry, I'm not entirely sure about the mechanism for forwarding an MSI from one core to another. 

According to the section 3.3 and 3.1.3.5 , the  specification mentions that the purpose of MSI.ppn is to direct an MSI to a guest interrupt file in an IMSIC. 

 In a bare-metal OS environment, which implies only stage1 translation exists, there is no guest interrupt file. I understand that after stage1 translation, the MSI can be directly routed to the IMSIC without msiptp enable.

ved-rivos, 2025-07-21:
Also please note that two stage address translation is always active though either stage can be effectively disabled by setting it Bare. There is always a GPA but the GPA may be same as the SPA if the G stage is Bare

ved-rivos, 2025-07-21:
> Since Stage-1 (s1) is not Bare and Stage-2 (s2) is Bare, why can't we directly map the virtual address to the target IMSIC interrupt file address through the s1 page tables? Why is an additional interrupt remapping entry needed to translate one SPA to another SPA?

It may not be very useful but the architecture supports it. Further, the capability allows use of MRIF for instance when a core has been offlined temporarily.

> If both Stage-1 (s1) and Stage-2 (s2) are bare mode, would we still need to determine whether an address targets an interrupt region? If a match occurs, would interrupt remapping entries still be required for translation?

The stage 1 and stage 2 are always active. There is no way to disable them. The address translation is always 
     IOVA -> S1 -> GPA -> S2 -> SPA

If stage 1 is bare then GPA == IOVA. If stage 2 is Bare then SPA == GPA. When MSI PTEs are enabled, the GPA which is the result of stage 1 is determine if it matches an address that should be translated using MSI PTEs and if it does then the MSI PTEs are used.

Chen-rui55, 2025-07-22:
Thank you, I've clarified the specification's descriptions of these features. The spec requires them to be relatively independent, and the specific usage depends on software . I think I still need to spend some more time diving into how software utilizes MSI.

closed 2025-07-22 by ved-rivos