riscv-non-isa/riscv-iommu #594: iommu_atc check

1252005989, 2025-07-23:
Hi,
If iohgatp.MODE is bare mode,  why do we still need to compare the GSCID?
And in this scenario, does the GSCID in the DDT make sense?

[image]

1252005989, 2025-07-23:
An example is as follows:
For two requests with different DIDs and PIDs but the same GSCID, PSCID, and request address, their stage1+stage2 modes are bare mode and they are in MSI flat mode (basic MSI). In this scenario, these two requests may have different MSI addresses, but when checked in the ATC, both yield the same result.

1252005989, 2025-07-23:
If iohgatp.MODE is bare mode(GV=0), why do we still need to compare the GSCID in ATC?

ved-rivos, 2025-07-23:
When iohgatp mode is bare the GSCID is not valid. When it’s not valid the model implementation fills the ATC with GV=gscid=0 . Your implementation can use any other method as needed to differentiate entries with valid and invalid GSCID

closed 2025-07-24 by ved-rivos