riscv-non-isa/riscv-iommu #601: SVNAPOT

asdvvcc123, 2025-07-31:
hi,sir
[image]

if gpte.N ==1 , the gst_page_sz need to change to 64KB?   and then the cache can hit it , but here size still is 4kb;

ved-rivos, 2025-07-31:
With Svnapot there are two address translation caching scheme that may be employed:

1. The address translation cache may either create a single TLB entry covering the entire NAPOT region.
2. The NAPOT PTEs may be supported by splitting the NAPOT region into TLB entries covering any smaller power-of-two region sizes. For example, a 64 KiB NAPOT PTE might trigger the creation of 16 standard 4 KiB TLB entries, all with contents generated from the NAPOT PTE (even if the PTEs for the other 4 KiB regions have different contents).

The model implements the second scheme. 

Irrespective of the scheme employed by the address translation caches, the result of the address translations and protection are identical.

asdvvcc123, 2025-07-31:
hi,sir: 
but in the code  no see this part ,  a 64 KiB NAPOT PTE might trigger the creation of 16 standard 4 KiB TLB entries; 
just change the  page_size to napot format and then into the tlb;

closed 2025-07-31 by asdvvcc123